gpiozero/gpiozero #829: Add sample_rate parameter for distancesensor

hyansuper, 2020-01-19:
I see a sample_wait parameter for the father class of distancesensor，I recommend adding it or the constructor of distancesensor

bennuttall, 2020-01-19:
Do you mean `sample_rate` or `sample_wait`?

lurch, 2020-01-20:
I just had a look through https://github.com/gpiozero/gpiozero/blob/master/gpiozero/input_devices.py and `SmoothedInputDevice` has a `sample_wait` parameter https://gpiozero.readthedocs.io/en/stable/api_input.html#smoothedinputdevice

For the `SmoothedInputDevice` sub-classes:
 * `LineSensor` has a `sample_rate` parameter which it passes to `SmoothedInputDevice` as `sample_wait=1 / sample_rate`
 * `MotionSensor` has a `sample_rate` parameter which it passes to `SmoothedInputDevice` as `sample_wait=1 / sample_rate`
 * `LightSensor` hardcodes `sample_wait=0.0`
 * `DistanceSensor` hardcodes `sample_wait=0.06`

No, I don't know why these all work slightly differently :wink:  ping @waveform80 

@hyansuper AFAICT a hardcoded `sample_wait=0.06` means that you can't get readings faster than 16 times a second? Did you open this issue because you need a faster update rate?

hyansuper, 2020-01-29:
Hi, @bennuttall, I mean sample_rate
I have multiple devices(DistanceSensors on multiple toy cars), so I want a lower sample_rate to reduce traffic on network, although it may not help.
but I notice that LineSensor has a  sample_rate param, so I think why not DistanceSensor has a sample_rate param too?

lurch, 2020-01-29:
Compared to the amount of data sent by a camera over the network, I suspect the amount of data sent by a DistanceSensor will be very minimal?

hyansuper, 2020-01-30:
Yes, you're probably right. I should try reduce camera framerate instead.
a more valid reason to request adding the parameter for `DistanceSensor` is consistency across `SmoothedInputDevice`s