gpiozero/gpiozero #834: multiple changes of led.blink() leads to segmentation fault

heizie, 2020-02-03:
Operating system:  Raspbian Buster
Python version: 3.7
Pi model: Pi zero
GPIO Zero version: 1.5.1
Pin factory used: RPiGPIO

If this led.blink() be implemented multiple times. It'll occur Segmentation fault.
There should be something to change the parameters for blinking?

pseudocode
`
def func_a():
    led.blinkt(0.1,0.1,2)
    return

def func_b():
    led.blinkt(0.1,0.1,3)
    return

while True:
    if 10 second after:
        func_a
    if somthings wrong:
        func_b`

bennuttall, 2020-02-03:
I don't follow. Can you post actual reproducable code so we can take a look?

heizie, 2020-02-03:
> I don't follow. Can you post actual reproducible code so we can take a look?

The problem is, when i implement the "led.blink()" in a loop (means, this function is runned multiple times). Something in the Raspberry Pi goes wrong. And then the programm stop running, output "Segmentation fault"

bennuttall, 2020-02-03:
Yeah, when you call `blink()` with `background=True` (the default) it spawns a new thread (after killing the old one), so it's no surprise that it causes problems.

This is not something we need to address in the library. You just need to add a sleep in your code, or rewrite to avoid constantly starting new blinks.

closed 2020-02-03 by bennuttall

lurch, 2020-02-03:
> There should be something to change the parameters for blinking?

There is. Did you look at the [documentation](https://gpiozero.readthedocs.io/en/stable/api_output.html#gpiozero.LED.blink)? :wink:

heizie, 2020-02-03:
> > There should be something to change the parameters for blinking?
> 
> There is. Did you look at the [documentation](https://gpiozero.readthedocs.io/en/stable/api_output.html#gpiozero.LED.blink)?

Thx, i've read that. But I'm searching for something, which can "push" the new params to the function., in order to avoid killing and then starting a new thread.
Because in my script, the blinking frequency should be changed dynamically (the params updated, while running the loop. the updated params are also generated in the loop).

heizie, 2020-02-03:
this is my script
```

from gpiozero import LED,Button,Buzzer
import datetime
import gspread
from oauth2client.service_account import ServiceAccountCredentials
import pandas as pd
import requests
import os

offed_flag = 0

led = LED(20)
bz = Buzzer(24)
l_button = Button(21)
r_button = Button(23)

def recon_push(status,gc):
    global offed_flag
    if status == 1 and offed_flag == 1:
        off_readed = pd.read_csv(''something not relevant'')
        off_list = off_readed.values.tolist()
        i = len(off_list)
        for i in range(0,i):
            wks = gc.open("clean-in-time").sheet1
            length = len(wks.col_values(1))
            wks.update_cell(length+1,1,off_list[i][0])
            wks.update_cell(length+1,2,off_list[i][1]+' lost connection')
            off_readed = off_readed.drop([i])
        off_readed.to_csv('something not relevant')
        offed_flag = 0
    return

def connect_check():
    pss = 'I wont tell you'
    b_conn = 'ifdown --force wlan0'
    r_conn = 'ifup wlan0'
    try:
        r = requests.get('http://www.google.com')
        status = 1
        led.blink(0.01,background = False)
        return status
    except:
        try:
            os.system('echo %s|sudo -S %s' % (pss, b_conn))
            os.system('echo %s|sudo -S %s' % (pss, r_conn))
            time.sleep(60)
            requests.get('http://www.google.com')
        except:
            status = 0
            #led.blink(0.8)
        return status

def gconnect():
    scope = ['https://spreadsheets.google.com/feeds',
            'https://www.googleapis.com/auth/drive']
    credentials = ServiceAccountCredentials.from_json_keyfile_name('some adress', scope)
    gc = gspread.authorize(credentials)
    return gc

def write_time(status, gc ,stat_name):
    global offed_flag
    time_now = str(datetime.datetime.now())
    if status == 1:
        gc.login()
        wks = gc.open("clean-in-time").sheet1
        status = 1
        length = len(wks.col_values(1))
        wks.update_cell(length+1,1,time_now)
        wks.update_cell(length+1,2,stat_name)
        bz.beep(0.1,0.1,3)
        #led.blink(0.01)

    elif status == 0:
        d1 ={'time':[time_now],'status':[stat_name]}
        df = pd.DataFrame(d1)
        df.to_csv(''something not relevant'',mode='a',header=False,index=False)
        bz.beep(1,1,3)
        #led.blink(0.8)
        offed_flag = 1
    return

def start_pressed():
    bz.beep(0.1,0.1,1)
    status = connect_check()
    write_time(status, gc,'start')
    return

def end_pressed():
    bz.beep(0.1,0.1,1)
    status = connect_check()
    write_time(status, gc,'end')
    return

if __name__ == "__main__":
    gc = gconnect()
    status = connect_check()
    t1 = datetime.datetime.now()
    l_button.when_pressed = start_pressed
    r_button.when_pressed = end_pressed
    while True:
        t2 = datetime.datetime.now()
        if int((t2 - t1).seconds) > 1:
            status = connect_check()
            recon_push(status, gc)
            t1 = datetime.datetime.now()

```

lurch, 2020-02-03:
Why mess about with `if int((t2 - t1).seconds) > 1:` ? Why not just do `time.sleep(1)` ?

> But I'm searching for something, which can "push" the new params to the function led.blink(), in order to avoid killing and then starting a new thread.

Thanks, now it makes more sense what you're trying to do. I don't think the blink_thread lets you do that though.

heizie, 2020-02-04:
> Why mess about with `if int((t2 - t1).seconds) > 1:` ? Why not just do `time.sleep(1)` ?

That's true!

But we still need something for pushing? Because the error which i'm now facing at, is about "killing and then starting a new thread" too offen. and leads to "segmentation error"

lurch, 2020-02-04:
Have you tried different [pin backends](https://gpiozero.readthedocs.io/en/stable/api_pins.html) ? It would be interesting to know if the segmentation fault is coming from the underlying pin library or from GpioZero itself.