gpiozero/gpiozero #837: RuntimeError: Not running on a RPi! with Ubuntu for Raspberry Pi

davidcorbin, 2020-02-13:
Operating system: 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=19.10
DISTRIB_CODENAME=eoan
DISTRIB_DESCRIPTION="Ubuntu 19.10"
Python version: e.g. 3.7
Pi model: e.g. Pi 4 Model B
GPIO Zero version: e.g. 1.5.1
Pin factory used: e.g. RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

Please give us as much information about your issue as possible. Write your
code inside code blocks like so:

```bash
ubuntu@ubuntu:~$ python3
Python 3.7.5 (default, Nov 20 2019, 09:21:52)
[GCC 9.2.1 20191008] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero import LED
>>> from time import sleep
>>> led = LED(9)
Traceback (most recent call last):
  File "/home/ubuntu/.local/lib/python3.7/site-packages/gpiozero/pins/pi.py", line 109, in pin
    pin = self.pins[n]
KeyError: 9

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ubuntu/.local/lib/python3.7/site-packages/gpiozero/devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/home/ubuntu/.local/lib/python3.7/site-packages/gpiozero/output_devices.py", line 211, in __init__
    pin, active_high, initial_value, pin_factory=pin_factory
  File "/home/ubuntu/.local/lib/python3.7/site-packages/gpiozero/output_devices.py", line 93, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/home/ubuntu/.local/lib/python3.7/site-packages/gpiozero/mixins.py", line 106, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/home/ubuntu/.local/lib/python3.7/site-packages/gpiozero/devices.py", line 521, in __init__
    pin = self.pin_factory.pin(pin)
  File "/home/ubuntu/.local/lib/python3.7/site-packages/gpiozero/pins/pi.py", line 111, in pin
    pin = self.pin_class(self, n)
  File "/home/ubuntu/.local/lib/python3.7/site-packages/gpiozero/pins/rpigpio.py", line 132, in __init__
    GPIO.setup(self.number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Not running on a RPi!
>>>
```

bennuttall, 2020-02-13:
Looks like an RPi.GPIO-on-Ubuntu issue, not a gpiozero issue.

I assume RPi.GPIO on its own doesn't work? Can you confirm:

```python
from RPi import GPIO

GPIO.setup(9, GPIO.IN)
```

How did you install RPi.GPIO and gpiozero? apt or pip?

@waveform80 Is RPi.GPIO working in eoan?

davidcorbin, 2020-02-13:
No, RPi.GPIO doesn't work on its own.

```bash
ubuntu@ubuntu:~$ python3
Python 3.7.5 (default, Nov 20 2019, 09:21:52)
[GCC 9.2.1 20191008] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from RPi import GPIO
>>> GPIO.setup(9, GPIO.IN)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Not running on a RPi!
>>>
```

I installed both through pip

davidcorbin, 2020-02-17:
It seems that rebooting the PI resolved this issue. I'm not sure what else it could be.

closed 2020-02-17 by davidcorbin

25077667, 2020-07-02:
Hi, I have meet this problem too.
But, reboot doesn't solve this problem....

OS: Ubuntu server 20.04
on Raspberry pi 4, 4 GB version
(Also had installed python's GPIO, but in runtime it says "RuntimeError: Not running on a RPi!")

bennuttall, 2020-07-02:
Try installing rpi.gpio with pip:

```bash
sudo pip3 install rpi.gpio --upgrade
```

schauveau, 2020-07-21:
Check that your user is in the 'dialout' group.

    # ls -l /dev/ gpiomem
    crw-rw---- 1 root dialout 239, 0 Apr  1 17:23 /dev/gpiomem
    # groups 
    ubuntu adm dialout cdrom floppy sudo audio dip video plugdev netdev lxd

In theory, that should already be the case for the default ubuntu user but new users will not get it by default.

Though, there could be something else going on. I experienced the problem temporarily with the official package python3-gpiozero 1.4.1-1.2 in ubuntu with a user that was already in the group dialout and it disappeared after upgrading gpiozero using pip3. Strangely, it still worked after uninstalling with pip3.  Could that simply be a kernel problem like a kernel module not being properly loaded or udev not creating /dev/gpiomem? I'll check that if that happens again

.

MarcoAndrade26, 2020-07-22:
Has someone solved this issue?

schauveau, 2020-07-23:
@MarcoAndrade26  It works for me. Do you have a /dev/gpiomem device and if so are you in the dialout group? Also, it could be a good idea to update gpiozero (using pip3) since the ubuntu package is not aware of the RPi4.

schauveau, 2020-07-23:
This is probably irrelevant but I commented out i2c_arm and spi in /boot/firmware/syscfg.txt because I2C was conflicting with dtoverlay=gpio-shutdown to use GPIO3 for a power button.

mmwagner02, 2020-09-28:
FWIW, I found this thread while searching for a solution--except that I am actually running RaspberryPi OS (raspbian).  I'm 99% sure that my problem was that I was operating as a user I had created myself, as opposed to the builtin `pi` user. 
The technique which made my problem go away was to add my user account to all of the same groups as the `pi` user.  I'm sure I didn't really have to add _all_ the same groups, but it was easier than figuring out which groups were actually required.

lurch, 2020-09-28:
https://github.com/raspberrypi/documentation/blob/master/configuration/security.md#changing-your-username may be helpful?

bischou, 2020-10-05:
I made this, as su
cd /dev
chmod og+rwx gpio*

All problems seem so far away :-)

mrts, 2020-10-10:
@bischou's solution is good, but needs a little more details.

This is a permissions issue, your user does not have read-write access to `/dev/gpiomem`. Here's one way to give access:

1. change `/dev/gpiomem` group ownership to your user's group:
```sh
sudo chown root:$USER /dev/gpiomem
```
2. allow group read-write access to `/dev/gpiomem`:
```sh
sudo chmod g+rw /dev/gpiomem
```

lurch, 2020-10-10:
As an alternative to step 1 (change `/dev/gpiomem` group ownership to your user's group), you could instead add yourself to whichever group currently owns `/dev/gpiomem` :slightly_smiling_face:  (AFAIK this is the more flexible option, and is less likely to have side-effects than changing the group ownership of `/dev/gpiomem`? :shrug: ). Step 2 would still be needed.

mrts, 2020-10-11:
@lurch, by default `/dev/gpiomem` group ownership is *root* in Ubuntu, so albeit a good principle, it won't work. I updated my comment to make that point clear.

lurch, 2020-10-11:
Ahhh, thanks for the clarification :+1: 
(I've never run Ubuntu on Raspberry Pi myself - just one of the many things I never seem to have the time for)

lgg, 2020-11-15:
I faced the same problem.

Solution:
Provide correct access rights to gpio devices in `/dev/`. NOT 777!!!

Check this: https://sourceforge.net/p/raspberry-gpio-python/code/ci/default/tree/create_gpio_user_permissions.py#l23

yangdangfu, 2020-11-21:
run the script with **sudo** can resolve the problem. For me I used the command on the script main.py :

sudo python3 main.py

lurch, 2020-11-21:
Also, see https://github.com/gpiozero/gpiozero/issues/893#issuecomment-729999234 where I discovered that only Ubuntu 20.10 includes a version of RPi.GPIO that has fully-working Raspberry Pi 4 support. Versions of Ubuntu prior to this only ship older versions of RPi.GPIO.

lgg, 2020-11-25:
> run the script with **sudo** can resolve the problem. For me I used the command on the script main.py :
> 
> sudo python3 main.py

It is a very bad way and insecure. Using sudo is not for running usual scripts. You just need to set up correct access rights on `/dev/gpio*` devices.

oliver-fried, 2021-01-23:
> I made this, as su
> cd /dev
> chmod og+rwx gpio*
> 
> All problems seem so far away :-)

yes, this worked for me at least!!

traemand, 2021-05-19:
> https://github.com/raspberrypi/documentation/blob/master/configuration/security.md#changing-your-username may be helpful?

This, from above link, fixed the issue for me on rpi 4 running Ubuntu Mate 20.04. 

`sudo usermod -a -G adm,dialout,cdrom,sudo,audio,video,plugdev,games,users,input,netdev,gpio,i2c,spi <INSERT USERNAME HERE>`

ksanim, 2021-08-25:
> I made this, as su
> cd /dev
> chmod og+rwx gpio*
> 
> All problems seem so far away :-)

THIS SOLVED THE PROBLEM. this was a permission issue after all. If you wanna know what chmod means, read https://linuxize.com/post/chmod-command-in-linux/

HelloZorex, 2021-09-12:
> This is a permissions issue, your user does not have read-write access to `/dev/gpiomem`. Here's one way to give access:
> 
> 1. change `/dev/gpiomem` group ownership to your user's group:
> 
> ```shell
> sudo chown root:$USER /dev/gpiomem
> ```
> 
> 1. allow group read-write access to `/dev/gpiomem`:
> 
> ```shell
> sudo chmod g+rw /dev/gpiomem
> ```
> 
This worked for me, but it seems that after a reboot, I still get the same error and have to do it all over again.

lurch, 2021-09-12:
> I don't think the permission change is getting saved.

Yeah, on some (most?) Linux distros the files in `/dev/` are a virtual filesystem, which means they get 'created' each time you boot up (and hence the permission changes aren't persistent over a reboot). ping @waveform80 for his Ubuntu expertise.

schauveau, 2021-09-12:
The permission can be set at boot using a udev rule as shown here 
https://www.thegeekdiary.com/how-to-configure-device-file-owner-group-with-udev-rules/
However, messing with the permissions is usually a bad idea.
The device has write access for the group so a better approach is to proceed the other way around: keep the current group of /dev/gpiomem and add your user to that group.  On my rpi4, that group is `dialout` so
```bash
(bash) sudo adduser $USER dialout
```
You may have to reboot or logout & login before the user get access to the the group.  
 
PS: if the group of /dev/gpiomem is `root` on your system then ignore my advice and use a udev rule instead.

schauveau, 2021-09-12:
Or simply speaking, create a file /etc/udev/rules.d/99-gpiomem.rules containing a single line
```
KERNEL=="gpiomem", OWNER="root", GROUP="dialout"
```

you can test it with 
```bash
sudo udevadm trigger /dev/gpiomem
```

LabheshMehta0112, 2021-09-12:
> @bischou's solution is good, but needs a little more details.
> 
> This is a permissions issue, your user does not have read-write access to `/dev/gpiomem`. Here's one way to give access:
> 
> 1. change `/dev/gpiomem` group ownership to your user's group:
> ```sh
> sudo chown root:$USER /dev/gpiomem
> ```
> 2. allow group read-write access to `/dev/gpiomem`:
> ```sh
> sudo chmod g+rw /dev/gpiomem
> ```
> 
> Note that it would be better to add your user to the group that owns `/dev/gpiomem` instead of changing group ownership as @lurch notes below. However, by default `/dev/gpiomem` group ownership is *root* in Ubuntu and adding your user to the *root* group would expand your user privileges which you probably don't want. Hence the group ownership change.

This worked for me, Thankx for the explanation
Also thankx to @bischou

HelloZorex, 2021-09-13:
> Or simply speaking, create a file /etc/udev/rules.d/99-gpiomem.rules containing a single line
> 
> ```
> KERNEL=="gpiomem", OWNER="root", GROUP="dialout"
> ```
> 
> you can test it with
> 
> ```shell
> sudo udevadm trigger /dev/gpiomem
> ```

yeah mine ended up being "root", so this helped fix my problem. Thanks!