gpiozero/gpiozero #838: GPIO Pin turns off immidiately after function ends.

XenottaBit, 2020-02-16:
I have a python app that allows me to toggle GPIO pins using remote GPIOZERO. I have a function for each of the two buttons that allows me to turn a specified pin on or off. However, when I run the function to turn the pin on, it turns off right after. 
```
# Pin Factory Configuration
factory = PiGPIOFactory(host='192.168.2.153')
# Event
def pinenable():
    PinID = int(txt.get())
    PIN(PinID, pin_factory=factory).on()
    statusvar = ("Pim " + txt.get() + " Is Raised High")
    statuslbl.config(text=statusvar)
```
Note: It also changes the text of a tkinter label.

bennuttall, 2020-02-16:
I don't know what `PIN` is but assuming it's `OutputDevice` or similar, you need to init the device in global scope and run the `on` method within the function. If you create the device object within the function it will be garbage collected and the pin restored to its original state.

```python
factory = PiGPIOFactory(host='192.168.2.153')
device = OutputDevice(pin_id)

def pinenable():
    device.on()
    statusvar = ("Pim " + txt.get() + " Is Raised High")
    statuslbl.config(text=statusvar)
```

XenottaBit, 2020-02-16:
Oh. I have a line at the top that imports LED from GPIOZERO as PIN. That's what it is

bennuttall, 2020-02-16:
Ok

closed 2020-02-16 by bennuttall

XenottaBit, 2020-04-14:
Well, now I have another problem. Every time i run the `pinenable` function, GPIOZERO tells me that I have an invalid pin factory. Here's the updated code:
```py
# Pin Factory Configuration
factory = PiGPIOFactory(host='192.168.2.238')

# On event
def pinenable():
    PinID = int(txt.get())
    led = LED(PinID, factory)
    led.on()
    statusvar.set("Pim " + txt.get() + " Has Been Raised")
```
And here's the output:
```
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1550, in __call__
    return self.func(*args)
  File "gui.py", line 52, in pindisable
    led = LED(PinID, factory)
  File "/home/itai/.local/lib/python2.7/site-packages/gpiozero/devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/home/itai/.local/lib/python2.7/site-packages/gpiozero/output_devices.py", line 211, in __init__
    pin, active_high, initial_value, pin_factory=pin_factory
  File "/home/itai/.local/lib/python2.7/site-packages/gpiozero/output_devices.py", line 93, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/home/itai/.local/lib/python2.7/site-packages/gpiozero/mixins.py", line 106, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/home/itai/.local/lib/python2.7/site-packages/gpiozero/devices.py", line 512, in __init__
    super(GPIODevice, self).__init__(**kwargs)
  File "/home/itai/.local/lib/python2.7/site-packages/gpiozero/devices.py", line 243, in __init__
    Device.pin_factory = Device._default_pin_factory()
  File "/home/itai/.local/lib/python2.7/site-packages/gpiozero/devices.py", line 280, in _default_pin_factory
    raise BadPinFactory('Unable to load any default pin factory!')
BadPinFactory: Unable to load any default pin factory!```

bennuttall, 2020-04-15:
The pin factory has to be a keyword argument: `LED(pin, pin_factory=factory)` not `LED(pin, factory)`

lurch, 2020-04-15:
Also, you're still only creating the LED object _inside_ the `pinenable` function. As @bennuttall mentioned previously, this means that when the function ends, the `led` variable will go out of scope, and be garbage-collected; and when any GpioZero objects get garbage-collected (destroyed) they automatically get restored to their "original state" i.e. turned off.

You probably want something like:
```python3
# Pin Factory Configuration
factory = PiGPIOFactory(host='192.168.2.238')
leds = dict()

# On event
def pinenable():
    PinID = int(txt.get())
    leds[PinID] = LED(PinID, pin_factory=factory, initial_value=True)
    statusvar.set("Pim " + txt.get() + " Has Been Raised")
```
`leds` is now a dictionary at the global scope, and so `leds[PinID]` won't go out of scope when the `pinenable` function ends. (note that I've also used the `initial_value` parameter, to save you having to make a separate call to `leds[PinID].on()` )

XenottaBit, 2020-04-16:
Well, that works, but when I try to turn it off by setting `initial_value` to `False`, it says the pin is in use by gpiozero.LED

bennuttall, 2020-04-16:
You need to create your LED object outside of the function. You can't initialise an LED multiple times.

lurch, 2020-04-16:
If you want to turn it off again, you can either use
```python3
leds[PinID].off()
```
or
```python3
leds[PinID].value = False
```
https://gpiozero.readthedocs.io/en/stable/api_output.html#gpiozero.LED