gpiozero/gpiozero #841: Using with individually addressable RGB LED strips

alexwaibel, 2020-03-03:
Device name: Individually addressable RGB LEDs
Protocol: e.g. GPIO
URL of example product: https://www.amazon.com/ALITOVE-WS2812B-Individually-Addressable-Waterproof/dp/B00ZHB9M6A
Further information e.g. data sheet:

Is it currently possible to control a strip of individually addressable LEDs using this library? The kind I am working with have only one data pin on them, so I would think they would be compatible but I have not yet been able to get them to light up, even all one color (which is my intended use case anyway).

[Here's an example of this](https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/) which uses the neopixel library. I feel like this is perhaps already possible with this library and would love to hear from anyone else with a similar setup.

alexwaibel, 2020-03-03:
Ahh, I needed to search further. Looks like this is answered in #807. Short answer is not supported at this time.

closed 2020-03-03 by alexwaibel

bennuttall, 2020-03-03:
You can use APA102 pixels with gpiozero, though it's not provided by the library yet, there's some code here: https://github.com/ThePiHut/rgbxmastree/blob/master/tree.py