gpiozero/gpiozero #843: Button(22) causes 100% CPU usage

234Mike432, 2020-03-12:
Operating system: Raspbian GNU/Linux 9 (stretch)
Python version: 3.5.3
Pi model: Pi ZeroW 1.1
GPIO Zero version: 1.4.1
Pin factory used: AttributeError: type object 'Device' has no attribute '_default_pin_factory'
See http://rpf.io/gpzissue for information on how to find out these details

I was trying to use GPIO22 as an input but when running the code the CPU usage goes to 100% even with no other code in the program.  The CPU usage stays at 100% even after the program has terminated and Python has been closed.  The only recovery method I found was to re-boot the Pi.  I tried it with GPIO2 and did not have the problem.  I have not tried other pins yet.

```python
import time
import spidev
from gpiozero import LED, Button

ReadySig = Button(22)

print('Done')
```

bennuttall, 2020-03-12:
Does it happen when you don't import spidev?

I can't reproduce this on a Pi 3. Can't see what's special about GPIO22. Have you previously set that pin up for something else? It might be worth trying to replicate on a fresh SD card (ideally using Buster) with up-to-date gpiozero and RPi.GPIO.

234Mike432, 2020-03-12:
I tried with only the following code and get the same results
`from gpiozero import Button
ReadySig = Button(22)
`

lurch, 2020-03-13:
What other software have you got installed on your Pi? Looks like _maybe_ some other program is using GPIO22, and when GpioZero tries using GPIO22 as well, that other program is getting confused and using 100% CPU?
If you tun `top` in a terminal window, can you see what process is using all the CPU?

234Mike432, 2020-03-16:
I tested this again today and its working fine with pin 22.  I did a bunch of pin reconfirmations on my other device and may have had this accidentally tied to an input on the other device.  So input to input may have been causing a constant button press/release leading to 100% CPU usage.

closed 2020-03-16 by 234Mike432