gpiozero/gpiozero #844: PingServer - name 'devnull' is not defined

SteveAmor, 2020-03-14:
Operating system: Raspbian Buster (clean install)
Python version: 2.7.16 & 3.7.3
Pi model: Pi 3
GPIO Zero version: 1.5.1
Pin factory used: RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

Fails at https://github.com/gpiozero/gpiozero/blob/master/gpiozero/internal_devices.py#L125

Fix for >= Python3.3:

```python
stdout=subprocess.DEVNULL, stderr=subprocess.DEVNULL)
```

bennuttall, 2020-03-14:
No - use `PingServer` the same way you'd use `Button`. Create an instance, and ask for its value:

```python
>>> from gpiozero import PingServer
>>> google = PingServer("google.com")
>>> google.value
1
```

See the docs: https://gpiozero.readthedocs.io/en/stable/api_internal.html#pingserver

SteveAmor, 2020-03-14:
Thanks Ben

closed 2020-03-14 by SteveAmor