gpiozero/gpiozero #851: New internal device to get GPU temperature?

rick2047, 2020-04-16:
Device name: GPUTemperature
Protocol: Internal
URL of example product: NA
Further information e.g. data sheet: NA

Please provide example code for using this device with Python, using another library:
```
def get_gpu_tempfunc():
    """ Return GPU temperature as a character string"""
    res = os.popen('/opt/vc/bin/vcgencmd measure_temp').readline()
    return res.replace("temp=", "")
```

bennuttall, 2020-04-16:
Nice idea. We should do it using a file interface if possible, like we do for CPU temperature.

lurch, 2020-04-16:
I think there's actually only a single temperature sensor in the Pi's SoC? https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/overclocking.md#monitoring-core-temperature

rick2047, 2020-04-16:
@lurch I don't read it that way. I have been monitoring my CPU and GPU temperatures for a week now and they are always different values. A degree Celsius or so apart.

rick2047, 2020-04-16:
@bennuttall I am willing to spend a bit more time and even include frequency and voltage related information as described in this page

https://www.maketecheasier.com/finding-raspberry-pi-system-information/

I don't know why anyone will use these things, do you think it would be valuable anyways?

lurch, 2020-04-16:
ping @pellwell for clarification about "CPU" and "GPU" temperatures.

bennuttall, 2020-04-16:
@rick2047 would be nice to include more stuff like this. Take a look at how the other internal devices are implemented: https://github.com/gpiozero/gpiozero/blob/master/gpiozero/internal_devices.py