gpiozero/gpiozero #853: Can't connect to remote gpio server.

pcs3rd, 2020-04-20:
Operating system: e.g. Raspbian Buster
Python version: e.g. 3.7
Pi model: e.g. Pi 4 Model B
GPIO Zero version: e.g. 1.5.1
Pin factory used: remote gpio

This is my code.
```
from gpiozero import Button
from gpiozero.pins.pigpio import PiGPIOFactory
from signal import pause

factory = PiGPIOFactory(host='192.168.0.87')

#button = Button(2)
button = Button(0, pin_factory=factory)

if Pin0.is_pressed:
  print("Pin 0 is high")
pause()
```  
and this is how im evoking rpigpiod:
```  
sudo pigpiod -n localhost -n 192.168.0.62
```  
and this is what my python script mentioned earlier is returning:
```  
 Traceback (most recent call last):
  File "readstate.py", line 8, in <module>
    button = Button(0, pin_factory=factory)
  File "/home/pi/.local/lib/python2.7/site-packages/gpiozero/devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/gpiozero/input_devices.py", line 432, in __init__
    bounce_time=bounce_time, pin_factory=pin_factory)
  File "/home/pi/.local/lib/python2.7/site-packages/gpiozero/mixins.py", line 383, in __init__
    super(HoldMixin, self).__init__(*args, **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/gpiozero/input_devices.py", line 182, in __init__
    pin_factory=pin_factory)
  File "/home/pi/.local/lib/python2.7/site-packages/gpiozero/mixins.py", line 197, in __init__
    super(EventsMixin, self).__init__(*args, **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/gpiozero/input_devices.py", line 98, in __init__
    super(InputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/home/pi/.local/lib/python2.7/site-packages/gpiozero/devices.py", line 521, in __init__
    pin = self.pin_factory.pin(pin)
  File "/home/pi/.local/lib/python2.7/site-packages/gpiozero/pins/pi.py", line 111, in pin
    pin = self.pin_class(self, n)
  File "/home/pi/.local/lib/python2.7/site-packages/gpiozero/pins/pigpio.py", line 228, in __init__
    raise ValueError(e)
ValueError: 'no permission to update GPIO'
```
What is causing this? is this my issue? Why is it doing this? I can't really seem to find anything definitive through a google search.

e

bennuttall, 2020-04-20:
Have you had it working before, or not at all? Are you sure the IP addresses are correct?

Does it work when run from the same Pi? (without specifying host, just using localhost)

Can you try using `pigpio` on its own?

```python
import pigpio

pi = pigpio.pi('192.168.0.87')

pi.read(0)
```

pcs3rd, 2020-04-20:
I haven't been able to get it working until kind of now.  
with your code, The connection works, but the only value returned is ``` 1 ``` regardless of the state of the button ( Using ```gpio read 0``` returns both ``` 1 ``` and ```0```

bennuttall, 2020-04-20:
Does that match what happens when you run it locally, with any library? Best to check the GPIO reading the button actually works at the lowest level.

pcs3rd, 2020-04-20:
No, it appears to only be happening with pigpio when accessing it remotely.
Using both RPi.GPIO locally (no remote gpio) and sysfs ( I think its called that), I can get ```0``` and ```1```.

pcs3rd, 2020-04-20:
rpigpio uses bcm pin numbers, not wiring pi numbers :/

closed 2020-04-20 by pcs3rd

bennuttall, 2020-04-20:
Ahh. You mean **gpiozero uses BCM numbers** (by default). I totally should have checked why you were using GPIO pin 0 - it's ok to do so but [GPIO0 is BOARD number 27](https://pinout.xyz/pinout/pin27_gpio0#)

You can in fact use [WiringPi numbering](https://gpiozero.readthedocs.io/en/stable/recipes.html#pin-numbering) if you want:

```python
button = Button("WPI0")
```

Again, there's humour in the fact the issue boiled down to **GPIO0** not working 😆