gpiozero/gpiozero #854: PCF8575

Misiu, 2020-04-28:
Device name: e.g. Expander
Protocol: I2C
URL of example product: http://www.ti.com/product/PCF8574
https://www.openimpulse.com/blog/products-page/product-category/pcf8574-io-expansion-module/
Further information e.g. data sheet: http://www.ti.com/lit/ds/symlink/pcf8574.pdf

Libraries that support this extender: https://pypi.org/project/pcf8574/, https://pypi.org/project/i2crelay/

I'd like to ask for adding support for PCF8575 expander.
It is 8 bit I/O, so it can be used to control 8 devices (8 relay board) or as 8 inputs.
It also allows to set individual bits as input or output, so 4 bits can work as input and 4 as outputs.

lurch, 2020-04-28:
See also #199

Unfortunately there's no I2C support in GpioZero yet.

Misiu, 2020-04-29:
@lurch I've created this issue because there is a comment on #199 by @mgiaco. I don't want this to get lost.
This is similar to https://github.com/gpiozero/gpiozero/issues/625, but it got closed.
https://github.com/gpiozero/gpiozero/pull/651 aimed to add I2C support.

I hope this will get added someday :)