gpiozero/gpiozero #855: MCP48x2

martynwheeler, 2020-05-01:
Device name: e.g. DAC
Protocol: e.g. SPI
URL of example product:
Further information e.g. data sheet: http://ww1.microchip.com/downloads/en/devicedoc/20002249b.pdf

Please provide example code for using this device with Python, using another library:

```python

```
How would I go about using aDAC with gpiozero?  It seems like it should be fairly straightforward, but I am struggling to get my head around sending data via SPI using gpiozero. If anyone can point me in the right direction I’d be very grateful

ukBaz, 2020-05-02:
Is the following class not fairly close to what you are looking for?
https://github.com/gpiozero/gpiozero/blob/9cba4aa68d02ecdf459c63737a7cd3f6c0eb57ab/gpiozero/spi_devices.py#L276

It has 16-bits and data is in the lower bits. Unfortunately it is only 9-bit resolution. The MCP3 seems to be "right justified" on the used bits where the MCP4 is "left justified". This might be an issue reading the correct value and mean that you will have create an MCP4xxx class which will be very similar to MCP3xxx

Anyway, this is untested but I would think that the following might be a start in the right direction and at least give you a value out even if it is the wrong value.

```python
class MCP48x2(MCP3xxx):
    def _send(self):
        # MCP48x2 protocol looks like the following:
        #
        #     Byte        0        1
        #     ==== ======== ========
        #     Tx   DxGSxxxx xxxxxxxx
        #     Rx   xxxxRRRR RRRRRRRR for the 4822
        #     Rx   xxxxRRRR RRRRRRxx for the 4812
        #     Rx   xxxxRRRR RRRRxxxx for the 4802
        data = 0x0000
        dac_select = 1 # 1 = B 0 = A
        gain_select = 1
        shutdown = 1
        data |= dac_select << 15
        data |= gain_select << 13
        data |= shutdown << 12
        print(f'data: {data:016b}')
        # data: 1011000000000000
        return self._int_to_words(data)
```

I am sure one of the library developers will correct what I got wrong :-)

martynwheeler, 2020-05-02:
Thank you, I will give it a try and report back to you!

martynwheeler, 2020-05-02:
Thanks for your help.  I went a bit further and wrote a new class:
`
from gpiozero import SPIDevice
from gpiozero.exc import DeviceClosed, SPIBadChannel, SPIBadArgs, InputDeviceError

class AnalogOutputDevice(SPIDevice):
    """
    Represents an analog output device connected to SPI (serial interface).
    """
    def __init__(self, bits, initial_value=0, **spi_args):
        if bits is None:
            raise InputDeviceError('you must specify the bit resolution of the device')
        self._bits = bits
        self._value = initial_value
        super(AnalogOutputDevice, self).__init__(shared=True, **spi_args)

    @property
    def bits(self):
        """
        The bit-resolution of the device/channel.
        """
        return self._bits

    def _write(self):
        raise NotImplementedError

    @property
    def value(self):
        return self._value

    @value.setter
    def value(self, value):
        self._value = value
        self._write()
        

class MCP48x2(AnalogOutputDevice):
    """
    Extends :class:`AnalogOuputDevice` to implement an interface for all DAC
    chips with a protocol similar to the Microchip MCP48x2 series of devices.
    """

    def __init__(self, channel=0, bits=8, initial_value=0, gain=1, **spi_args):
        if not 0 <= channel < 2:
            raise SPIBadChannel('channel must be 0 or 1')
        self._channel = channel
        if not 0 <= gain < 2:
            raise InputDeviceError('gain must be 0 or 1')
        self._gain = gain
        if bits not in (8, 10, 12):
            raise InputDeviceError('bit resolution must be 8, 10, or 12')
        self._shutdown = 1
        super(MCP48x2, self).__init__(bits, initial_value, **spi_args)

    @property
    def channel(self):
        """
        The channel to write data to. The MCP48x2 have 2 channels (0-1).
        """
        return self._channel
        
    @property
    def gain(self):
        """
        The gain for the output. The MCP48x2 can have a gain of 1x or 2x.
        """
        return self._gain
        
    @property
    def value(self):
        return self._value

    @value.setter
    def value(self, value):
        if not 0 <= value < (2**self.bits):
            raise InputDeviceError('value must be in the range 0 to ' + str(2**self.bits))
        self._value = value
        self._shutdown = 1
        self._write()
        
    def shutdown(self):
        self._shutdown = 0
        self._write()
        
    def _write(self):
        self._words_to_int(
            self._spi.transfer(self._send())[-2:], self.bits
            )

    def _send(self):
        # MCP48x2 protocol looks like the following:
        #
        #     Byte        0        1
        #     ==== ======== ========
        #     Tx   DxGSRRRR RRRRRRRR for the 4822
        #     Tx   DxGSRRRR RRRRRRxx for the 4812
        #     Tx   DxGSRRRR RRRRxxxx for the 4802
        data = 0x0000
        data |= self.channel << 15
        data |= self.gain << 13
        data |= self._shutdown << 12
        data |= self.value << (12 - self.bits)
#        print(f'data: {data:016b}')
        return self._int_to_words(data)

`

Would it be worth adding this into the library?

lurch, 2020-05-04:
I don't have time to look through your code in detail, but one quick comment...
> class AnalogOutputDevice(SPIDevice):
    """
    Represents an analog output device connected to SPI (serial interface).
    """

What about if somebody wants to add an I2C-connected DAC chip in the future? Yes, I know that [`AnalogInputDevice` inherits directly from `SPIDevice`](https://gpiozero.readthedocs.io/en/stable/_modules/gpiozero/spi_devices.html#AnalogInputDevice) - IMHO that was a mistake too :wink:

martynwheeler, 2020-05-04:
Yes, that is a good point! Sent from Mail for Windows 10 From: Andrew SchellerSent: 04 May 2020 16:39To: gpiozero/gpiozeroCc: martynwheeler; AuthorSubject: Re: [gpiozero/gpiozero] MCP48x2 (#855) I don't have time to look through your code in detail, but one quick comment...class AnalogOutputDevice(SPIDevice):"""Represents an analog output device connected to SPI (serial interface)."""What about if somebody wants to add an I2C-connected DAC chip in the future? Yes, I know that AnalogInputDevice inherits directly from SPIDevice - IMHO that was a mistake too 😉

achjaj, 2023-08-29:
> Thanks for your help. I went a bit further and wrote a new class (details can be found [here](https://leicesterraspberrypi.wordpress.com/guides/using-gpiozero-with-a-digital-to-analog-converter/)).
> 
> ```
> from gpiozero import SPIDevice
> from gpiozero.exc import DeviceClosed, SPIBadChannel, SPIBadArgs, InputDeviceError
>  
> class AnalogOutputDevice(SPIDevice):
>    ...
>  
> class MCP48x2(AnalogOutputDevice):
>    ...
> ```
> 
> Would it be worth adding this into the library?

@martynwheeler can I use this code? I want to make an interface for MCP DACs, and maybe in the future I will create a pull request

martynwheeler, 2023-08-29:
Sure, no problem