gpiozero/gpiozero #856: Select leds in LEDBoard.blink and LEDBoard.pulse

anxuae, 2020-05-02:
Add a way to select the LEDs of a ``LEDBoard`` that we want to make ``blink`` or ``pulse``.

**Purpose of change:**

1. Starting to blink/pulse each LED one by one does not ensure that all the blinking/pulsing LEDs are synchronized. The synchronization is done by the ``blink_thread`` of the board, but for the moment, it is only possible to start/stop all LEDs.

2. This change will also make the API more consistent with ``on()`` and ``off()`` methods which offer the possibility to select the concerned LEDs.

**Example usage of gpiozero with suggested change:**

```python

board = LEDBoard(green=3, yellow=4, red=5)

board.blink(on_time=0.3, off_time=0.6, leds=(board.green, board.red))

```

bennuttall, 2020-05-03:
Sounds like a reasonable idea - not sure when we'll have time to work on it though.

An alternative is to set source:

```python
def blink_some(on_state):
    off state = (0, ) * len(on_state)
    while True:
        yield on_state
        yield off_state
    
leds = LEDBoard(2, 3, 4, 5, 6)

leds.source = blink_some((1, 0, 1, 0, 1, 0))
leds.source_delay = 1
```

This will blink odd leds every second.

anxuae, 2020-05-03:
Yes, it can work for simple sequence.

However for complex ones, it implies a large amount of code in the source generator.
To reduce the generator complexity, the LED state could be get from external variables (global variables, class attributes, ...) which are updated by the rest of the application. I'm not sure that setting variables during generator iteration is thread-safe?

lurch, 2020-05-04:
Of course an even simpler workaround would be to just set up multiple LEDBoards, with one LEDBoard for just the LEDs that you want to blink in unison, and a separate LEDBoard for the LEDs that you want to control individually :wink: 

Yes, this obviously removes a lot of flexibility, but I did say **simpler** workaround! :laughing:

anxuae, 2020-05-04:
Yes that's true, I will do that for now even if it is not intuitive😉.