gpiozero/gpiozero #857: ButtonBoard.when_pressed is called 2 times, ButtonBoard.when_released never called

anxuae, 2020-05-05:
Operating system: **Raspbian Buster**
Python version: **3.7**
Pi model: **Pi 3 Model B**
GPIO Zero version: **1.5.1**
Pin factory used: **RPiGPIO**

Using the class `ButtonBoard` to create a group of buttons, I noticed that the `when_pressed` callback is called 2 times: one the a button is pressed, an other when the same button is released. However, the `when_released` callback is never called.

This is easily reproducible on any OS/Platform using the `MockFactory` backend, see the code bellow:

```python
import time
from gpiozero import ButtonBoard, Device
from gpiozero.pins.mock import MockFactory

Device.pin_factory = MockFactory()
pin5=Device.pin_factory.pin(5)
pin6=Device.pin_factory.pin(6)

def when_pressed(b):
    print("PRESSED!")
    print("is_active=", b.is_active, "is_held=", b.is_held, "is_pressed=", b.is_pressed, "value=", b.value)
    print("active_time=", b.active_time, "inactive_time=", b.inactive_time, "pressed_time=", b.pressed_time, "held_time=", b.held_time)

def when_released(b):
    print("RELEASED!")
    print("is_active=", b.is_active, "is_held=", b.is_held, "is_pressed=", b.is_pressed, "value=", b.value)
    print("active_time=", b.active_time, "inactive_time=", b.inactive_time, "pressed_time=", b.pressed_time, "held_time=", b.held_time)

board=ButtonBoard(b1=5, b2=6, pull_up=True)
board.when_pressed = when_pressed
board.when_released = when_released

pin5.drive_low()
time.sleep(1)
pin5.drive_high()

pin6.drive_low()
time.sleep(1)
pin6.drive_high()
```

Gives the following output:

```
PRESSED!
is_active= True is_held= False is_pressed= True value= ButtonBoardValue(b1=1, b2=0)
active_time= 0.00022869899999999888 inactive_time= None pressed_time= 0.0002455959999999868 held_time= None
PRESSED!
is_active= False is_held= True is_pressed= False value= ButtonBoardValue(b1=0, b2=0)
active_time= 0.0005650759999999977 inactive_time= None pressed_time= 0.0005819090000001026 held_time= 0.00029694599999996463
PRESSED!
is_active= True is_held= True is_pressed= True value= ButtonBoardValue(b1=0, b2=1)
active_time= 0.001045900000000044 inactive_time= None pressed_time= 0.0011205370000000325 held_time= 0.0016530569999999578
PRESSED!
is_active= False is_held= True is_pressed= False value= ButtonBoardValue(b1=0, b2=0)
active_time= 0.0009639689999998424 inactive_time= None pressed_time= 0.0010001809999997668 held_time= 1.004356276
```

bennuttall, 2020-05-05:
I think this is the same as #761

anxuae, 2020-05-05:
Yes, that True, sorry for the duplicate.
I see that there is a PR under review on this topic, you can close it.

anxuae, 2020-05-05:
Or maybe, I can use it to explain what I'm try to achieve with the **ButtonBoard**, I have not found the solution yet, help would be appreciated 😉.

I have 3 functions: `func_a`, `func_b` and `func_c` and 2 buttons on my board `b1` and `b2`.

- `func_a` shall be called when `b1` is pressed but not `b2`
- `func_b`  shall be called when `b2` is pressed but not `b1`
- `func_c` shall be called when `b1` and `b2` are pressed

Considering that if the time difference `b1` pressed and `b2` pressed is less that 0.3s, means that both buttons are pressed together, thus **only ** `func_c` shall be called.

For the moment I have only succeed in calling `func_c` but in this case `func_a` and `func_b` are also called. Do you have an idea how to achieve that?

lurch, 2020-05-05:
I don't really think there's any way to achieve that cleanly in GpioZero. Maybe you want to look at using `when_held` instead of `when_pressed` (adjusting the hold-time as appropriate) to stop the "press events" on buttons b1 and b2 firing immediately?

bennuttall, 2020-05-05:
Can you do something like this?

```python
bb = ButtonBoard(2, 3)

def call_fn(b):
    value = tuple(b.value)
    fn = {
        (0, 0): None,
        (1, 0): func_a,
        (0, 1): func_b,
        (1, 1): func_c,
    }[value)
    if fn:
        fn()

bb.when_pressed = call_fn
```

anxuae, 2020-05-05:
After try, this is exactly the issue that I'm facing to.

It is physically impossible to press 2 buttons at same time. Thus with this method, generally ``func_a`` or ``func_b`` are called, sometime ``func_c`` but after others.

lurch, 2020-05-05:
What about _something_ like:
```python3
b1 = Button(2, hold_time=0.5)
b2 = Button(3, hold_time=0.5)

def held1():
    if b2.is_pressed:
        # b1 was held while b2 was pressed
        func_c()
    else:
        # b1 was held but b2 not pressed
        func_a()

def held2():
    if b1.is_pressed:
        # b2 was held while b1 was pressed
        # but don't do anything here, let held1 handle it instead
        pass
    else:
        # b2 was held but b1 not pressed
        func_b()

b1.when_held = held1
b2.when_held = held2
```
That's entirely untested though, so may need tweaking or it may not work at all :wink:

bennuttall, 2020-05-05:
Somewhat related, though specific to `Button`: https://gpiozero.readthedocs.io/en/latest/faq.html#how-do-i-use-button-when-pressed-and-button-when-held-together

anxuae, 2020-05-06:
Ho yeah, this is the solution! It's working perfectly.
And instead of setting a `bounce_time`, I adjust the bounce time using `hold_time`.

Maybe this solution could be kept in the FAQ, or imagine a specific callback for ButtonBoard to catch "simultaneous" button press.

Thanks a lot! This issue can be closed.

lurch, 2020-05-06:
> Maybe this solution could be kept in the FAQ

I dunno, I imagine for most use-cases (and the kind of beginner audience that GpioZero is aimed at) it'll just be easier to stick with one button per function? :stuck_out_tongue_winking_eye:

bennuttall, 2021-02-22:
Closing in favour of #761

closed 2021-02-22 by bennuttall

waveform80, 2021-03-21:
This is a harder problem than most people realize (unless you've dug into keyboard firmwares before :). Personally I find it easiest to figure these things out by sketching out the possible waveforms and seeing what patterns we're looking for. Anyway, while I don't think this belongs in the documentation (well, maybe in the advanced recipes chapter), here's my little solution:

```python3
from gpiozero import Button
from signal import pause
from time import time

btn1 = Button(26)
btn2 = Button(19)

times = {
    # Set up a lookup table of btns and pressed states, mapped times last seen
    (btn, pressed): time() - 1
    for btn in (btn1, btn2)
    for pressed in (False, True)
}

# There's several patterns we're concerned with, which the following pattern
# of waveforms illustrate:
#
# btn1     ,----.            ,-------.          ,------.      ,---.
#          |    |            |       |          |      |      |   |
#      ----'    `------------'       `----------'      `------'   `--------
#               :                                      :
# btn2          :  ,-----.       ,------.   ,-------.  :   ,---------.
#               :  |     |       |      |   |       |  :   |         |
#      ------------'     `-------'      `---'       `------'         `-----
#               :        :              :              :             :
#             btn1     btn2           both           both          both
#             press    press          press          press         press
#
# We'll record the times of all presses and releases but we're only going to
# *do* something when both buttons are released. At that point we'll compare
# the times of activation and decide whether the two activations coincided.

def btn_pressed(btn):
    times[btn, True] = time()

def btn_released(btn):
    times[btn, False] = time()
    other = btn2 if btn is btn1 else btn1
    if not other.is_pressed:
        if times[btn, True] < times[other, True] < times[btn, False]:
            print('Both pressed')
        elif times[other, True] < times[btn, True] < times[other, False]:
            print('Both pressed')
        elif btn is btn1:
            print('btn1 pressed')
        else:
            print('btn2 pressed')

btn1.when_pressed = btn2.when_pressed = btn_pressed
btn1.when_released = btn2.when_released = btn_released
pause()
```