gpiozero/gpiozero #859: Energenie with pi model B

martynwheeler, 2020-05-10:
Operating system:. Raspbian Buster
Python version:  3.7
Pi model:  Pi v1 Model B
GPIO Zero version: 1.5.1
Pin factory used: e.g. RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

Is ir possible to use gpiozero with energenie on a model b (pi v1)? 

Thank you

lurch, 2020-05-10:
I think the only workaround for now will be to edit your local version of https://github.com/gpiozero/gpiozero/blob/master/gpiozero/boards.py#L1993 and update it to use the correct GPIO numbers for your old Raspberry Pi.
Alternatively, you could just buy a Raspberry Pi Zero WH ? :wink:

bennuttall, 2020-05-10:
Can you try checking out this branch: https://github.com/gpiozero/gpiozero/pull/860

bennuttall, 2021-03-12:
The fix for this has been merged, pending a release (very soon)

closed 2021-03-12 by bennuttall