gpiozero/gpiozero #861: Remote pin factory is not set for Robot/Motor classes

jbernardo3, 2020-05-25:
When running Remote GPIO Recipe 5.5 (Remote Button Robot) as written (minus the specfic ip address of the remote machine) the remote pin factory appears to not be set because a button wired to the local machine does not affect the pin output on the remote machine. (Specifically using a pair of raspberry pi 4B's).

Running Remote GPIO Recipe 5.1 (LED + Button) as written (minus the specfic ip address of the remote machine) works as expected with the same setup. 

Confirmed both results with a multimeter.

jbernardo3, 2020-05-25:
I forgot to note that I tested the same pin that works with 5.1 using a Robot and Motor object and I got the same disparate result.

bennuttall, 2020-05-25:
This has been fixed but not released. You'll have to install from master:

```bash
sudo pip3 install https://bennuttall.com/files/gpiozero-1.5.2b1-py2.py3-none-any.whl
```

Should be released fairly soon

jbernardo3, 2020-05-26:
Sorry I didn't see this was a duplicate issue. Thanks a lot! This has been frustrating me for longer than I care to admit...

closed 2020-05-26 by jbernardo3

lurch, 2020-05-26:
Just for reference (IMHO cross-links are always useful) this was fixed in #792 .