gpiozero/gpiozero #862: Pinout Command Line Tool not included in python2 install.

urbanvisuals, 2020-05-27:
Hi.

This isn't much of an issue, but I had only installed the python2 libraries, and couldn't find the pinout command line tool. Maybe the doc page for the tool could say that it's only installed with the Python 3 install, or add it to the Python 2 installer, too. 

Thanks,

Operating system: e.g. Raspbian Buster
Python version: e.g. 2.7. 3.5.3
Pi model: e.g. Pi 3 Model B+
GPIO Zero version: e.g. 1.5.1

lurch, 2020-05-27:
IIRC it's only in the Python3 package, because you'd get file-conflicts if it was also included in the Python2 package.
Hopefully you're aware that Python2 is now EOL, and you ought to be using Python3 for all new development? :wink: 

> Maybe the doc page for the tool could say that it's only installed with the Python 3 install

Good suggestion!

bennuttall, 2020-05-27:
+1 to that

commit 9952332c (2021-02-22): Add note about pinout in the python 2 debian package, close #862
closed 2021-02-22 by bennuttall