gpiozero/gpiozero #863: documentation edit suggestion regarding cleanup

ViennaMike, 2020-05-27:
The Migrating from RPi.GPIO documentation currently states: 
Cleanup
Pin state cleanup is explicit in RPi.GPIO, and is done manually with GPIO.cleanup() but in GPIO Zero, cleanup is automatically performed on every pin used, at the end of the script. Manual cleanup is possible by use of the :meth:`~Device.close` method on the device.

Read more in the relevant FAQ: :ref:`gpio-cleanup`

I suggest modifying this to make it clear that this is NOT automatic upon abnormal termination. Propose rewriting (EMPHASIS ADDED to show differences as:

Cleanup
Pin state cleanup is explicit in RPi.GPIO, and is done manually with GPIO.cleanup() but in GPIO Zero, cleanup is automatically performed on every pin used, at the end of NORMAL TERMINATION OF the script. Manual cleanup is possible by use of the :meth:`~Device.close` method on the device. MANUAL CLEANUP MAY BE DESIRABLE IN TRY / EXCEPT / FINALLY BLOCKS TO CLEAR THE PINS UPON ABNORMAL TERMINATION, INCLUDING KEYBOARD INTERRUPTS. 

Read more in the relevant FAQ: :ref:`gpio-cleanup`

I would propose the same change in Sections 9.7 and 10.5 of the docs.

lurch, 2020-05-28:
Oh, I didn't realise that GpioZero _didn't_ cleanup upon abnormal exit.
Should it @bennuttall ?  (relates indirectly to #707 ?)

bennuttall, 2020-05-29:
I had this conversation with @waveform80 once - he thought it should too - but no, it doesn't.

If it's fixable that would be good. But for now / otherwise, I agree with adding clarity and a try/except to the docs.

lurch, 2020-05-29:
> If it's fixable that would be good.

Is there already an issue for that? If not perhaps there should be :)

bennuttall, 2020-05-29:
No - I didn't consider it a bug. Is it?

lurch, 2020-05-29:
Well it does seem like surprising (inconsistent) behaviour that pins that get otherwise auto-cleaned-up on script-exit, **don't** get auto-cleaned-up if the script exits by throwing an error.
IMHO, naturally :wink:

ViennaMike, 2020-05-29:
I'm a complete newcomer to GPIO Zero, so that's the context I'm coming from, but it does strike me as inconsistent and it would be better to either always automatically cleanup if that is not too hard a change. 
Background: I came across this while developing some new software that uses GPIO Zero's Button to receive input from a PIR sensor to trigger a prop, and AngleServo to drive a servo. The software, when in use, runs constantly, so I frequently used a keyboard interrupt to stop the software during development. 
It seems good practice to clean up however one's program terminates, and if one adds code to do that then the fact that the cleanup occurs automatically upon normal termination doesn't really help much. So my two cents is modify the documentation in the short term, but do add a new issue to handle abnormal terminations as well.

lurch, 2020-05-31:
I just noticed that https://github.com/gpiozero/gpiozero/issues/323#issuecomment-218543961 also implies that "unintentional script termination" should perform device cleanup? :wink: 

ping @waveform80

bsimmo, 2020-07-14:
As an addition to this.
If I set LED on, they stay on when the program finishes.  This is in Thonny and Mu
If I use the Terminal and python3 xxx.py  it turns them all off.

Never noticed before as I tend to use Thonny or Mu when teaching/pi jams

They do not behave the same.

Also could there be an option to force not to clean up (so for terminal users currently) since there is one to force clean up.
(i.e. someone may want to briefly run a script and leave an led on, rather than a self contained program)
Make it all a bit more consistent.

closed 2021-02-22 by bennuttall

waveform80, 2021-03-20:
From reading the above it seems like people are mistaking one thing for another. The `atexit` handlers that gpiozero uses to clean up on script termination *always* run, whether termination is normal or abnormal. However, what we refer to as termination is the shutdown of the *Python interpreter running your script*, because that's when `atexit` runs.

However, what may be confusing people is that if you are running under Thonny, Mu, IDLE, or other things that embed Python interpreters then if your script crashes with an exception the Python interpreter is *still running*. Most of these environments work this way so that you can have a "live environment" in which to perform post-mortem analysis. In other words, the statement that's added to the documentation should state that this issue is specific to those embedded environments: it doesn't happen if you're simply running your script "normally".

I'd be interested to know if there's a way to detect if we're running under an embedded environment; all the info I've seen so far has involved horrible hacks like checking for tell-tale modules in `sys.modules` or walking the framestack looking for known things. Urgh.

commit bdfb40ce (2021-02-22): Merge pull request #925 from gpiozero/863-cleanup-docs

Explain how to cleanup after exception, close #863
commit c271d70f (2021-02-22): Explain how to cleanup after exception, close #863