gpiozero/gpiozero #864: pinout: show memory in GB when it's >=1GB

bennuttall, 2020-05-28:
Since Pi 4, Pis now have multiple GB of RAM so using Mb is obtuse.

When you run `pinout` on an GB Pi 4, it shows 1024Mb, 2048Mb, 4096Mb, 8192Mb. Would be better to show 1GB/2GB/4GB/8GB.

It shouldn't change in `pi_info()` but should change in `pinout` output.

lurch, 2020-05-28:
Also, doesn't `Mb` with a lower-case `b` typically indicate Mega-bits, with `MB` being used to indicate Mega-Bytes? :man_shrugging:

bennuttall, 2020-05-28:
Yeah I thought that

lurch, 2020-05-28:
Hopefully we won't get anybody asking us to use `GiB` :wink:

drewrisinger, 2020-12-21:
I think this can be closed. This is the current result of pinout (v1.5.1), on a raspberry Pi 4B 4GB (RAM is shown in GB):

```
$ pinout
,--------------------------------.
| oooooooooooooooooooo J8   +======
| 1ooooooooooooooooooo  PoE |   Net
|  Wi                    oo +======
|  Fi  Pi Model 4B  V1.2 oo      |
|        ,----.               +====
| |D|    |SoC |               |USB3
| |S|    |    |               +====
| |I|    `----'                  |
|                   |C|       +====
|                   |S|       |USB2
| pwr   |HD|   |HD| |I||A|    +====
`-| |---|MI|---|MI|----|V|-------'

Revision           : c03112
SoC                : BCM2711
RAM                : 4.0GB
Storage            : MicroSD
USB ports          : 4 (excluding power)
Ethernet ports     : 1
Wi-fi              : True
Bluetooth          : True
Camera ports (CSI) : 1
Display ports (DSI): 1

J8:
   3V3  (1) (2)  5V
 GPIO2  (3) (4)  5V
 GPIO3  (5) (6)  GND
 GPIO4  (7) (8)  GPIO14
   GND  (9) (10) GPIO15
GPIO17 (11) (12) GPIO18
GPIO27 (13) (14) GND
GPIO22 (15) (16) GPIO23
   3V3 (17) (18) GPIO24
GPIO10 (19) (20) GND
 GPIO9 (21) (22) GPIO25
GPIO11 (23) (24) GPIO8
   GND (25) (26) GPIO7
 GPIO0 (27) (28) GPIO1
 GPIO5 (29) (30) GND
 GPIO6 (31) (32) GPIO12
GPIO13 (33) (34) GND
GPIO19 (35) (36) GPIO16
GPIO26 (37) (38) GPIO20
   GND (39) (40) GPIO21
```

lurch, 2020-12-21:
Yeah, fixed by #892

closed 2021-03-19 by waveform80