gpiozero/gpiozero #866: Lost contact with relay

JeroenFree, 2020-06-01:
Operating system: e.g. Raspbian Buster
Python version: e.g. 3.7
Pi model: e.g. Pi 4 Model B
GPIO Zero version: e.g. 1.5.1
Pin factory used: e.g. RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

Please give us as much information about your issue as possible. Write your
code inside code blocks like so:

```python

print("code here")
when i read status remote pi

then the pins short on or of when i read.

Led.is_lit or value
so relays is clicking of and on

also if initial = set none

again

```
greets doomer

lurch, 2020-06-01:
Huh? Could you try explaining that in a bit more detail please?

(I wonder if you're describing #707 ?)

JeroenFree, 2020-06-02:
Yes u paste code after work

Verzonden vanaf mijn Xperia van Sony-smartphone

---- Andrew Scheller schreef ----

JeroenFree, 2020-06-02:
#!/usr/bin/python
#reading a smb180 on a samba share :) <<<
#
import datetime
import sys
import io

# gpio to gpiozero  <<<
#local pins
#import RPi.GPIO as GPIO
import gpiozero
from gpiozero import *
from gpiozero import LED
from gpiozero import CPUTemperature

from gpiozero.pins.pigpio import PiGPIOFactory
factory4 = PiGPIOFactory(host='192.168.1.134')

led = LED(17, initial_value=None ,  active_high=True,  pin_factory=factory4)

class TemperatureSensor(CPUTemperature):
    @property
    def temperature1(self):
        with io.open(self.sensor_file, 'r') as f:
            return round(float(f.readlines()[1].split("=")[1]) / 1000 , 1)

temp = TemperatureSensor('/mnt/pi2/sys/bus/w1/devices/28-020f92455856/w1_slave')

import smbus as smbus
import time
from ctypes import c_short

DEVICE = 0x77

bus = smbus.SMBus(1)

def convertToString(data):
  return str((data[1] + (256 * data[0])) / 1.2)

def getShort(data, index):
  return c_short((data[index] << 8) + data[index + 1]).value

def getUshort(data, index):
  return (data[index] << 8) + data[index + 1]
  REG_CALIB  = 0xAA
  REG_MEAS   = 0xF4
  REG_MSB    = 0xF6
  REG_LSB    = 0xF7
  CRV_TEMP   = 0x2E
  CRV_PRES   = 0x34
  OVERSAMPLE = 3

  cal = bus.read_i2c_block_data(addr, REG_CALIB, 22)

  AC1 = getShort(cal, 0)
  AC2 = getShort(cal, 2)
  AC3 = getShort(cal, 4)
  AC4 = getUshort(cal, 6)
  AC5 = getUshort(cal, 8)
  AC6 = getUshort(cal, 10)
  B1  = getShort(cal, 12)
  B2  = getShort(cal, 14)
  MB  = getShort(cal, 16)
  MC  = getShort(cal, 18)
  MD  = getShort(cal, 20)

  # Lees Temperatuur
  bus.write_byte_data(addr, REG_MEAS, CRV_TEMP)
  time.sleep(0.005)
  (msb, lsb) = bus.read_i2c_block_data(addr, REG_MSB, 2)
  UT = (msb << 8) + lsb

  # Lees Luchtdruk
  bus.write_byte_data(addr, REG_MEAS, CRV_PRES + (OVERSAMPLE << 6))
  time.sleep(0.04)
  (msb, lsb, xsb) = bus.read_i2c_block_data(addr, REG_MSB, 3)
  UP = ((msb << 16) + (lsb << 8) + xsb) >> (8 - OVERSAMPLE)

  # Opmaak temperatuur
  X1 = ((UT - AC6) * AC5) >> 15
  X2 = (MC << 11) / (X1 + MD)
  B5 = X1 + X2
  temperature = (B5 + 8) >> 4
  # Opmaak luchtdruk
  B6  = B5 - 4000
  B62 = B6 * B6 >> 12
  X1  = (B2 * B62) >> 11
  X2  = AC2 * B6 >> 11
  X3  = X1 + X2
  B3  = (((AC1 * 4 + X3) << OVERSAMPLE) + 2) >> 2

  X1 = AC3 * B6 >> 13
  X2 = (B1 * B62) >> 16
  X3 = ((X1 + X2) + 2) >> 2
  B4 = (AC4 * (X3 + 32768)) >> 15
  B7 = (UP - B3) * (50000 >> OVERSAMPLE)

  P = (B7 * 2) / B4

  X1 = (P >> 8) * (P >> 8)
  X1 = (X1 * 3038) >> 16
  X2 = (-7357 * P) >> 16
  pressure = P + ((X1 + X2 + 3791) >> 4)

  return (temperature/10.0,pressure/ 100.0)

def main():
#here is my problem
#when i read status from led on or off 
#then the pomp = 220 relay card is set to True so the relay lost power from the gpio on remote pi 
#this is a fraction of time so when i add a 22 uf cap then te problem is not existing

#remember hot = off

 if led.is_lit == False:

        deg = u'\xb0'
        (temperature,pressure)=readBmp180()

        print "Zwembad Pomp :  Staat aan"
        print "Temperatuur  : ", temperature,deg.encode('utf8'), "C"
        print "Luchtdruk    : ", pressure, "mbar"
        print "Zembad water : ", temp.temperature1,deg.encode('utf8'), "C"

        f = open("/var/scripts/text.php","w")

        print >>f, '<?php'
        print >>f, '$temp = " Temperatuur       : ', temperature,deg.encode('utf8'),'";'
        print >>f, '$mbar = " Luchtdruk mbar  : ', pressure, '";'
        print >>f, '$grond = "Zwembad water  : ', temp.temperature1,deg.encode('utf8'),'";'
        print >>f, '$pomp = " Pomp staat" ;'
        print >>f, '$power = " aan ";'
        print >>f, '$stat = aan;'

        print >>f, '?>'

        f.close()

if led.is_lit == True:

     deg = u'\xb0'
        (temperature,pressure)=readBmp180()

        print "Zwembad Pomp :  Staat uit"
        print "Temperatuur  : ", temperature,deg.encode('utf8'), "C"
        print "Luchtdruk    : ", pressure, "mbar"
        print "Zembad water : ", temp.temperature1,deg.encode('utf8'), "C"

        f = open("/var/scripts/text.php","w")

        print >>f, '<?php'
        print >>f, '$temp = " Temperatuur       : ', temperature,deg.encode('utf8'),'";'
        print >>f, '$mbar = " Luchtdruk mbar  : ', pressure, '";'
        print >>f, '$grond = "Zwembad water  : ', temp.temperature1,deg.encode('utf8'),'";'
        print >>f, '$pomp = " Pomp staat " ;'
        print >>f, '$power = " uit ";'
        print >>f, '$stat = uit;'
        print >>f, '?>'

        f.close()

if __name__=="__main__":

   main()

lurch, 2020-06-02:
Hi @JeroenFree ,

Your code may "seem obvious" to you, but to an outside observer it's pretty unintelligible, so it's impossible to know what might be causing your problem :confused: 
Could you please try slimming it down to a _much_ smaller reproducible test-case? Thanks!

JeroenFree, 2020-06-02:
#!/usr/bin/python
#Importeer bibliotheek voor systeemfuncties.
import datetime
import sys
import io

#local pins
#import RPi.GPIO as GPIO
import gpiozero
from gpiozero import *
from gpiozero import LED
from gpiozero.pins.pigpio import PiGPIOFactory
factory4 = PiGPIOFactory(host='192.168.1.134')

led = LED(17, initial_value=None , pin_factory=factory4)

def main():

        print led.is_active  <<< here is the problem

if __name__=="__main__":

   main()

lurch, 2020-06-02:
Right, so it probably **is** the same issue as #707 that I guessed at earlier then, where the pin gets "turned off" automatically when GpioZero closes the pin (which happens when the Python script exits).

So I'd guess your solution is to either use pigpio directly and bypass GpioZero; or modify your script to be a long-running loop (so that the pin never gets closed) rather than treating it as a "single shot" script.

P.S. If I'm understanding what you're _actually_ doing, I think
```python
relay_state = DigitalInputDevice(17, pin_factory=factory4)
```
would be less confusing than
```python
led = LED(17, initial_value=None , pin_factory=factory4)
```
 ?

closed 2020-07-17 by bennuttall