gpiozero/gpiozero #869: Ratio doc is wrong

bwduncan, 2020-06-16:
https://github.com/gpiozero/gpiozero/blob/07fbc5f242294eedb52b9b331d2a905c2635e67e/gpiozero/input_devices.py#L784

A 2:3 ratio would give a 2/5ths voltage divider, I think what's meant here is a 1:2 ratio.

lurch, 2020-06-16:
Having just read the surrounding paragraph, it looks like it's referring to the output_voltage:total_voltage ratio (rather than the lower_resistance:upper_resistance ratio). And indeed 5*(2/3) results in the 3.333V mentioned.

But I'm only a software developer, not an electrical engineer, so I've no idea how these things are normally described :shrug:

bwduncan, 2020-06-16:
Fair enough, I think the example makes it pretty clear, so anyone trying it will hopefully get the right answer. I'm not an engineer either, and physicists and electrical engineers have a long history of disagreement anyway. From a quick look at some online resources, it seems that ratios are a commonly-confused tool, so I'm happy just to close this. Thanks for your time!

closed 2020-06-16 by bennuttall