gpiozero/gpiozero #872: Questions - ButtonBoard

BKeyport, 2020-06-25:
Is there effective tutorials for how ButtonBoard is used? 

I'm not sure how to pull out one button press from the many in that case. 

Also, is there a support board somewhere for GPIOZero? I'd rather be posting this in a forum type environment rather than as a "issue" so that the many users can help out. If there is, can you add that to the README.md? 

Thanks!

lurch, 2020-06-26:
> Is there effective tutorials for how ButtonBoard is used?

I think https://gpiozero.readthedocs.io/en/stable/api_boards.html#buttonboard is the best you currently have.

> I'm not sure how to pull out one button press from the many in that case.

Looking at https://github.com/gpiozero/gpiozero/blob/master/tests/test_boards.py#L107 I think you'd just do e.g. `bboard[2].is_pressed` ?

BKeyport, 2020-06-28:
Lurch: playing with it a bit, and I've got something going, but sadly, gpiozero is spelling everything out, so I might as well go back to not using boards at all. 

I get results like "ButtonBoardValue(device_0=0, device_1=1, device_2=0, device_3=0, device_4=0, device_5=0, device_6=0, device_7=0)" when it would be better to simply have "0100000" which I could then use effectively. 

Oh well.. Until it's documented properly, I'll just spin it out the way I had it before. Was trying to re-write a script for expansion, figured I'd make it "easier" but as always, when there's an easy way to do it, it'll fail.

closed 2020-06-28 by BKeyport

bennuttall, 2020-06-28:
Rather than `ButtonBoardValue(device_0=0,...` you just want to wrap `tuple` around your value:

```
In [3]: bb = ButtonBoard(*range(10))                                            

In [4]: bb.value                                                                
Out[4]: ButtonBoardValue(device_0=0, device_1=0, device_2=0, device_3=0, device_4=0, device_5=0, device_6=0, device_7=0, device_8=0, device_9=0)

In [5]: tuple(bb.value)                                                         
Out[5]: (0, 0, 0, 0, 0, 0, 0, 0, 0, 0)
```

I wrote an example similar to this for someone else a while ago, which gave you a decimal representation of the "binary value" of the button board (`Number` here is an extension of `ButtonBoard`):

```
>>> n = Number(2, 3, 4)
>>> n.number
0
# check it again while a button is pressed
>>> n.number
2
# check it again while another button is pressed
>>> n.number
6
```

See implementation here: https://github.com/gpiozero/gpiozero/issues/839

> Oh well.. Until it's documented properly, I'll just spin it out the way I had it before. Was trying to re-write a script for expansion, figured I'd make it "easier" but as always, when there's an easy way to do it, it'll fail.

Can you let us know what you expected to find?