gpiozero/gpiozero #873: Add active_high property to Motor class

sajjadaemmi, 2020-07-13:
Add active_high (bool) property to Motor class

Purpose of change:
many relay modules are active_low. this causes led.off() turn led on and led.on() turn led off. this problem can be solved by set active_high parameter to False in LED object.
but Motor has not active_high property and it makes very problems to work with relay and Motor.

```python
motor = Motor(forward=17, backward=18, active_high =False)
```

lurch, 2020-07-13:
> it makes very problems to work with relay and Motor.

Could you show some example code? I _think_ the Motor class assumes you're using a motor driver chip rather than a relay? See e.g. https://gpiozero.readthedocs.io/en/stable/recipes.html#motors

sajjadaemmi, 2020-07-13:
As I said, the relay inverse the low and high. for example when I write 
`motor.stop()`
 both forward and backward turn on

lurch, 2020-07-13:
If all you're doing is switching a relay on and off, could you use the `DigitalOutputDevice` class instead of a `Motor` class? :man_shrugging: 
https://gpiozero.readthedocs.io/en/stable/api_output.html#digitaloutputdevice

The only time I go anywhere near 220V is if I'm using https://gpiozero.readthedocs.io/en/stable/api_boards.html#energenie :wink:

sajjadaemmi, 2020-07-14:
i implemented own Motor class

```
from gpiozero import DigitalOutputDevice

class Motor:
    def __init__(self, forward, backward, active_high=True, initial_value=False, pin_factory=None):
        self.forward_motor = DigitalOutputDevice(pin=forward, active_high=active_high)
        self.backward_motor = DigitalOutputDevice(pin=backward, active_high=active_high)

    def forward(self):
        self.forward_motor.on()
        self.backward_motor.off()

    def backward(self):
        self.forward_motor.off()
        self.backward_motor.on()

    def stop(self):
        self.forward_motor.off()
        self.backward_motor.off()
    
    def close(self):
        self.forward_motor.close()
        self.backward_motor.close()

```

and It works as I like.
i suggest you add this option. because many developers need it.
thanks

lurch, 2020-07-14:
>  It works as I like.

That's great to hear! :tada: 
But I think it'll be "safer" if you change `forward` to
```
def forward(self):
    self.backward_motor.off()
    self.forward_motor.on()
```
because with your current code, if you did `motor.backward()` and then `motor.forward()` there'd be a fraction of a second where `backward_motor` and `forward_motor` would both be `on()` at the same time, and that's probably a bad idea when 220V is involved! :wink: 

> because many developers need it.

Oh? AFAIK you're the first person to ask for this? ping @bennuttall @waveform80

bennuttall, 2020-07-14:
Am I right in thinking that swapping the forward and backward pins would solve this?

lurch, 2020-07-14:
No, that'd only change the direction that the motor spins in :wink: