gpiozero/gpiozero #879: Attempting to use GPIO35 kills wifi (Pi Zero W)

ThiagoVinicius, 2020-07-26:
Operating system: Raspbian Stretch
Python version: 3.5.3
Pi model: Pi Zero W
GPIO Zero version: 1.5.0
Pin factory used: gpiozero.pins.rpigpio.RPiGPIOFactory

Full detail:
```python
Python 3.5.3 (default, Jul  9 2020, 13:00:10)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpiozero
>>> gpiozero.pi_info()
PiBoardInfo(revision='9000c1', model='Zero W', pcb_revision='1.1', released='2017Q1', soc='BCM2835', manufacturer='Sony', memory=512, storage='MicroSD', usb=1, ethernet=0, wifi=True, bluetooth=True, csi=1,dsi=0, headers=..., board=...)
>>> gpiozero.Device._default_pin_factory()
<gpiozero.pins.rpigpio.RPiGPIOFactory object at 0xb69ff610>
>>> import pkg_resources
>>> pkg_resources.require('gpiozero')[0].version
'1.5.0'
```

Following the docs about [Controlling the Pi’s own LEDs](https://gpiozero.readthedocs.io/en/stable/recipes_advanced.html#controlling-the-pi-s-own-leds) kills wifi connectivity on my device. It needs to be rebooted before it works again. The Pi itself appears to continue to run just fine, except for the missing wifi. While looking for similar bugs, I came across [raspberrypi/linux/issues/1342](https://github.com/raspberrypi/linux/issues/1342) and, sure, I do have that problem as well, with or without GPIO Zero.

I'm not exactly sure if this bug belongs here or there. I chose to report it here, where it has the most surprising effect.

Onto the report:

*led.py* is as follows:
```python
import gpiozero

power = gpiozero.LED(35, active_high=False)
print(power)
print('done')
```

```
pi@raspberrypi:~ $ echo none | sudo tee /sys/class/leds/led0/trigger
none
pi@raspberrypi:~ $ echo gpio | sudo tee /sys/class/leds/led1/trigger
tee: /sys/class/leds/led1/trigger: No such file or directory
gpio
pi@raspberrypi:~ $ python3 led.py > led.out 2> led.err
```

Wifi is dead right now. There's no more output. No new SSH connections get through. No ping receives answers.

After a reboot (that I had scheduled in advance), it is possible to see that the program executed to the end, only complaining about GPIO35 not being a physical pin (maybe it isn't the led?):

```
pi@raspberrypi:~ $ cat led.out
<gpiozero.LED object on pin GPIO35, active_high=False, is_active=False>
done
pi@raspberrypi:~ $ cat led.err
/usr/lib/python3/dist-packages/gpiozero/pins/pi.py:270: PinNonPhysical: no physical pins exist for GPIO35
  'no physical pins exist for %s' % repr(self)))
pi@raspberrypi:~ $
```

[excerpt of kern.log, from led.py execution until system reboot](https://github.com/gpiozero/gpiozero/files/4978516/kern.log)

I have no idea what trying to mess with this LED has to do with wifi, but it is 100% reproductible here. Running this script always kills wifi.

As of today, I'm unwilling to update Raspbian to Buster or try the newer wifi firmware. The reason being that this Pi is effectively headless (I have lost the only mini-hdmi adapter I had) -- If I kill the network for real, I'll be locked out of it.

I'm posting this issue as I'm curious to see if someone else is able to reproduce it, and maybe shed some light on what's going on.

ThiagoVinicius, 2020-07-26:
For one, I've just discovered that the only led on this Pi Zero W seems to be the activity Led, connected to GPIO47 (and on **/sys/class/leds/led0**). Accessing it causes no issues.

OTOH, docs might need to account for that. I expected to be controlling the non-existing power LED via the existing **/sys/class/leds/led0.

What is hooked to GPIO35, on that board, anyway?

bennuttall, 2020-07-26:
I've seen this before. I think it's the I2C bus which links the Wifi. I know on the Pi 3 the on-board LEDs moved to an I2C expander meaning they can't be controlled this way. Maybe the Pi Zero is the same, I can't remember. You might want to look around or ask on the forums.

lurch, 2020-07-27:
> What is hooked to GPIO35, on that board, anyway?

According to https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/bcm2835-rpi-zero-w.dts nothing is connected to GPIO35. Any ideas @pelwell ?

pelwell, 2020-07-27:
`raspi-gpio get 35` will show that it's the SDIO CMD line to the WiFi/BT combo chip - that's on 34-39. QED.

lurch, 2020-07-27:
Does that mean that the DTS file is misleading? Or am I just misunderstanding what the info in the DTS is representing?

pelwell, 2020-07-27:
The (upstream) DTS file is wrong - that part has probably just not been updated since the Zero.

ThiagoVinicius, 2020-07-27:
No wonder, then, that messing with GPIO35 would cause issues with wifi.

I'll leave the issue open, in case you want to update the docs to explicitly say what GPIO actually goes to the activity LED on the Pi Zero W.

Of course you may close it, if further action is unwarranted.

lurch, 2020-07-27:
https://gpiozero.readthedocs.io/en/stable/recipes_advanced.html#controlling-the-pi-s-own-leds already says "On the Pi Zero you can control the activity LED with this recipe, but there’s no separate power LED to control (it’s also worth noting the activity LED is active low, so set active_high=False when constructing your LED component)." ?

ThiagoVinicius, 2020-07-27:
It was the piece about GPIO47 being **/sys/class/leds/led1**. But that was me *reading it wrong*. It is correct on the docs. There's no issue left.

Thanks.

closed 2020-07-27 by ThiagoVinicius