gpiozero/gpiozero #881: Add Long/Short Buttons

ozone89, 2020-08-10:
* **Short description of change**: implement long/short button press in API

* **Where would this change be made**: Button class

* **Purpose of change**:
Add an option to the Button class for providing long/short pressing (in my example, by introducing a `dual_func` bool).  
At current state, if you try to invoke a function by holding a button, any other function called by `when_pressed` or `when_released` is executed as well, unless comparisons with `hold_time` are introduced in the code

* **Please provide example usage of gpiozero with your suggested change**: 

```python
define shortPress():
    print("short press")

define shortRelease():
    print("short press released")

define longPress():
    print("long press")

define longRelease():
    print("long press release")

button = Button(17, hold_time=5, dual_func=True)

button.when_pressed = shortPress
button.when_released = shortRelease

button.when_held = longPress
button.when_held_released = longRelease
```

lurch, 2020-08-10:
Does https://gpiozero.readthedocs.io/en/stable/faq.html#how-do-i-use-button-when-pressed-and-button-when-held-together help?

ozone89, 2020-08-10:
ah!
definitively didn't knew about such a workaround!

thanks and sorry for the inconvenience! :)

lurch, 2020-08-10:
Can this issue be closed then?

ozone89, 2020-08-10:
Ah, yes, sorry, forgot to close "^_^
Could it be still possible to consider the suggestion? :)

closed 2020-08-10 by ozone89