gpiozero/gpiozero #883: Documentation on how to create new device classes

gunthercox, 2020-08-15:
The threaded design for the device classes provided in this library seems to provide a really great development experience. However, I would like to be able to easily create new device classes for various sensors and hardware that I've built or purchased. A bit of guidance on how to do so would be greatly appreciated, and could possibly improve the existing documentation.

**Where would this change be made:** The project's documentation. Or, as a comment on this ticket if there is a simple solution that I have missed :slightly_smiling_face:

**Purpose of change:** This change would provide guidelines on how to create new devices to help developers write code for custom input and output devices.

Feel free to close this ticket off if you don't deem it to be constructive or necessary. If there is any interest in this change, I would also be happy to help update the documentation if I was pointed in the right direction.

bennuttall, 2020-08-15:
What kind of devices did you have in mind? As long as they're GPIO or SPI you should be able to. If they're another protocol like I2C (we don't support that yet), the best you can do is replicate a _similar_ API.

The best way to learn how to create gpiozero devices is to study the docs and the source code. I'd recommend reading (in this order):

- https://gpiozero.readthedocs.io/en/latest/recipes.html
- https://gpiozero.readthedocs.io/en/latest/recipes_advanced.html
- https://gpiozero.readthedocs.io/en/latest/api_generic.html
- https://gpiozero.readthedocs.io/en/latest/api_pins.html

It's not vital to understand all the pin factories, just know how the pin model works. Using the default pin factory is fine, but it's also worth knowing about mock pins.

Then pick a device from the right hand side of [this diagram](https://gpiozero.readthedocs.io/en/latest/_images/device_hierarchy.svg), find it in the source code, read it, read thPiSte comments, then study its parent class, and repeat all the way up the hierarchy. Start with things like PiStop, SnowPi, MotionSensor, and follow them up the tree. Look at the internal devices which are similar but not related to pins. Look at composite devices like LEDBoard, ButtonBoard and some HATs. Look at more Motor, Robot, Servo, TonalBuzzer.

It would be good practice to make your own composite devices. Say you have a few LEDs and buttons. Pretend they're a HAT and make a class for the board, making the API work like other HATs. Try something more complex, like combining motors with LEDs and buttons.

If you have a particular device in mind that we don't support (and it's pure GPIO), take a look at the issues here, to see if it's been suggested or mentioned before. Any comments from me, @waveform80 and @lurch will likely shed some light. If you're able to get started, find the class of the device it's most similar to, and try to design the API based on that. Which class should it extend? What methods and properties should it provide? How would it work in conjunction with other classes?

It's also worth studying the test suite: https://github.com/gpiozero/gpiozero/tree/master/tests

If you think there's a way to describe to others how to do this in the docs, I'd love to hear your ideas.

Hope that helps. Let me know if you have any questions.

gunthercox, 2020-08-16:
Thank you for this information, this seems like an excellent starting place and it definitely helps. Yes, I2C is actually the majority of the devices that I would be writing code for (motor controllers, sensor hats, etc.).

I'll let you know if I come up with any suggestions for ways to describe this.