gpiozero/gpiozero #885: PIGPIO_ADDR=192.168.1.3 python3

CTECH9, 2020-08-24:
Short description of change: Make Windows-specific environment variables for remote GPIO
Where would this change be made: e.g. Chapter 4.3

Purpose of change:

The environment variables provided do not work with the Windows 10 version of python, they are for Unix-based operating systems.

waveform80, 2021-03-15:
PRs welcome?

CTECH9, 2021-04-13:
I am dumb. I didn't know what I was doing and I didn't read the entire documentation. You may ridicule me as much as you like. I'm going to close the issue.

closed 2021-04-13 by CTECH9

waveform80, 2021-04-13:
Heh, no problem :) I did wonder if I was missing something as I *thought* it should "just work" under Windows, but unfortunately I haven't got any Windows machines around here at the moment to test on. Anyway, glad to hear it's sorted!