gpiozero/gpiozero #891: Add info on Docker setup

Mean-Street, 2020-09-21:
Hi,

I was looking for a way to run gpiozero inside a Docker container on a RaspPi, didn't find the info on how to allow access to GPIO devices here. Found a solution that I share and that might be added to the README:

2 methods:
* add the `/dev/gpiomem` device to the container:
    * through `docker run`: add `--device=/dev/gpiomem:/dev/gpiomem`
    * through Docker compose: add the following to your `docker-compose.yml`

```
        devices:
            - /dev/gpiomem:/dev/gpiomem
```

* run in privileged mode (`docker run --privileged`): see [docker run privilege documentation](https://docs.docker.com/engine/reference/run/#runtime-privilege-and-linux-capabilities). Not the preferred way from security point of view as it grants nearly full host access to the Docker process

Hope it helps!

lurch, 2020-09-21:
Just out of curiosity, what's the use-case for being able to run gpiozero inside a container? (I've never run Docker on a Pi)

Mean-Street, 2020-09-22:
I have a bunch of services running on my Pi (Samba server, Pi-Hole, OpenHAB, Telegram bot) that I dockerise and deploy with a `docker-compose.yml` file because it's easier to manage and make services interact. 
The Telegram bot allows to interact with all this and I wanted an option to get system info (temperature, disk usage, CPU usage...) returned, and it's where I use gpiozero

lurch, 2020-09-22:
> I wanted an option to get system info (temperature, disk usage, CPU usage...) returned

I don't _think_ any of those should require access to `/dev/gpiomem` ?  (If they do, maybe that's a bug?)

To clarify my earlier Docker use-case query - I'm not sure how or if things might go wrong if gpiozero is being used simultaneously from within two different containers? Question for @waveform80 probably.

Mean-Street, 2020-09-22:
I did some more tests:

* when I run `docker build -t gpiozerotest . && docker run -it gpiozero test`:

```
Successfully tagged gpiozerotest:latest
/usr/local/lib/python3.6/dist-packages/gpiozero/devices.py:279: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python3.6/dist-packages/gpiozero/devices.py:279: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python3.6/dist-packages/gpiozero/devices.py:279: PinFactoryFallback: Falling back from pigpio: No module named 'pigpio'
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python3.6/dist-packages/gpiozero/devices.py:279: PinFactoryFallback: Falling back from native: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root
  'Falling back from %s: %s' % (name, str(e))))
Traceback (most recent call last):
  File "/main.py", line 10, in <module>
    log.info("Temperature: %.1f \xb0C\n" % CPUTemperature().temperature)
  File "/usr/local/lib/python3.6/dist-packages/gpiozero/devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/gpiozero/internal_devices.py", line 179, in __init__
    super(CPUTemperature, self).__init__(pin_factory=pin_factory)
  File "/usr/local/lib/python3.6/dist-packages/gpiozero/mixins.py", line 197, in __init__
    super(EventsMixin, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/gpiozero/devices.py", line 243, in __init__
    Device.pin_factory = Device._default_pin_factory()
  File "/usr/local/lib/python3.6/dist-packages/gpiozero/devices.py", line 280, in _default_pin_factory
    raise BadPinFactory('Unable to load any default pin factory!')
gpiozero.exc.BadPinFactory: Unable to load any default pin factory!
```

* when I run `docker build -t gpiozerotest . && docker run -it --device /dev/gpiomem gpiozero test`:

```
Successfully tagged gpiozerotest:latest
/usr/local/lib/python3.6/dist-packages/gpiozero/devices.py:279: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python3.6/dist-packages/gpiozero/devices.py:279: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
/usr/local/lib/python3.6/dist-packages/gpiozero/devices.py:279: PinFactoryFallback: Falling back from pigpio: No module named 'pigpio'
  'Falling back from %s: %s' % (name, str(e))))
INFO:root:Temperature: 55.3 \xb0C

INFO:root:Load average: 1.4 \xb0C

INFO:root:Temperature: 24.6 \xb0C
```

But this seems to happen only when it falls back to the native pin factory. When I add `pip3 install RPi.GPIO` in the Dockerfile, it works even without the `--device /dev/gpiomem` parameter (and obviously without the warnings):

```
Successfully tagged gpiozerotest:latest
INFO:root:Temperature: 52.1 \xb0C

INFO:root:Load average: 0.5 \xb0C

INFO:root:Temperature: 24.6 \xb0C
```

Here's `Dockerfile`: 
```
FROM ubuntu:18.04

RUN apt-get update && apt-get install -y libsystemd-dev python3.5 python3-dev python3-pip
RUN pip3 install pystemd gpiozero

COPY "./main.py" "/main.py"
CMD ["python3", "/main.py"]
```

And `main.py`:

```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import logging as log
log.basicConfig(level=log.INFO)
from gpiozero import CPUTemperature, LoadAverage, DiskUsage

if __name__=="__main__":
    log.info("Temperature: %.1f °C\n" % CPUTemperature().temperature)
    log.info("Load average: %.1f °C\n" % LoadAverage().load_average)
    log.info("Temperature: %.1f °C\n" % DiskUsage().usage)
```

Hope it helps. Sorry I'm not very familiar with pin factories

waveform80, 2021-03-15:
The docker info *might* be useful to add to the docs (I'm skeptical, but hey - it can't harm), but I have to say that using gpiozero just to get CPU temperature, load-average, and disk usage info is ... bonkers!

You'd be much better off just cat'ing some files than loading an entire framework that queries what Pi you're running on and iterates through a list to try and load an underlying library to control a whole sub-system you're not even using :)

* Temperature: cat /sys/class/thermal/thermal_zone0/temp (divide by 1000)
* Load average: cat /proc/loadavg (three values: 1, 5, 15 minute averages)
* Disk usage: df /

cloudwitch, 2022-01-23:
Reason for me on this is I like to build everything in containers so that it's portable.  I'm trying to write an API that interacts with the GPIO and not having luck with it so far.

lurch, 2022-01-23:
As already mentioned I have very little Docker experience, but I know that e.g. Balena uses containers very extensively, so perhaps https://www.balena.io/docs/learn/develop/hardware/ might give you some pointers in the right direction? :man_shrugging:  
The best approach would probably be to get "just" `RPi.GPIO` working, and then don't start experimenting with `GPIO Zero` until after you've got over that initial hurdle?