gpiozero/gpiozero #893: is_pressed always returns true

MinePro120, 2020-10-02:
Operating system: Ubuntu server 20.04.1 64 bit
Python version: 3.8.2
Pi model: Pi 4 Model B
GPIO Zero version: 1.4.1-1.2
Pin factory used: Device._default_pin_factory() returns "AttributeError: type object 'Device' has no attribute '_default_pin_factory'"

Consider the following script :

```python
#!/bin/python3

from gpiozero import Button
import time
import os

stopButton = Button (26)

while True:
     if stopButton.is_pressed:
         os.system ("echo hi")
     time.sleep (1)
exit (1)
```
, it always prints "hi", no matter if GPIO 26 is set to ground or not. It's a simplified version of the script found [here](https://core-electronics.com.au/tutorials/how-to-make-a-safe-shutdown-button-for-raspberry-pi.html). It has worked fine form me on Raspberry Pi OS a couple of days ago, before I installed Ubuntu.

lurch, 2020-10-02:
ping @waveform80 as this sounds like an Ubuntu-specific issue.

bennuttall, 2020-10-02:
`Device._default_pin_factory() returns "AttributeError: type object 'Device' has no attribute '_default_pin_factory'"` is because it's an old version of gpiozero. `stopButton.pin_factory` should show the one used. Have you tried installing the latest version though, using pip to install?

As @lurch said, this isn't a gpiozero issue, more an Ubuntu thing. But it would be good to know which pin factory is being used. If it's `Native` then try installing RPi.GPIO too.

MinePro120, 2020-10-03:
I found out the Raspberry Pi OS ships with the same version which has worked for me before. I already tried installing the latest version using pip and running the script but then weird things happened; not only did I get a bunch of errors, but also even when the script terminated, the pi became unresponsive (or at least sshd that I'm using to connect) so much so I had to cut the power. I'll give it one more shot.

MinePro120, 2020-10-14:
I figured out RPi.GPIO works unreliably as well. So I'll report on Ubuntu and then here if anything persists.

MinePro120, 2020-11-18:
It seems the internal pull-ups are not working. Does anyone have any idea why? The i2c module is not listed with lsmod and I've already tested GPIOs 26 and 17.

lurch, 2020-11-18:
#900 fixes the internal pull-ups on the Pi4 if you're using NativePin.

MinePro120, 2020-11-18:
Is NativePin another factory (I'm using RPi.GPIO)? Anyways, I installed gpiozero and RPi.GPIO using pip3 and everything works fine now.

closed 2020-11-18 by MinePro120

lurch, 2020-11-18:
> Is NativePin another factory

Yup https://gpiozero.readthedocs.io/en/stable/api_pins.html#changing-the-pin-factory

Ahhh, good to hear that you've managed to solve your problems with Ubuntu :+1: 

I just did a quick bit of investigation, and the version of RPi.GPIO in PyPI (as used by `pip`) is 0.7.0 https://pypi.org/project/RPi.GPIO/ with one of the listed changes being "Fixed pull up/down for Pi4". And the version of RPi.GPIO used by Ubuntu 20.04 is 0.6.5 https://packages.ubuntu.com/focal/python3-rpi.gpio :facepalm: 
It looks like only Ubuntu 20.10 includes the version 0.7.0 needed for full Pi4 GPIO functionality https://packages.ubuntu.com/groovy/python3-rpi.gpio

MinePro120, 2020-11-19:
When the LTS is less stable... Thanks for the help!