gpiozero/gpiozero #895: Pull-down issue with `NativeFactory` on a Pi 4

boukeas, 2020-10-12:
Operating system: Raspbian Buster
Python version: 3.7
Pi model: tested on Pi 3 and 4 Model B
GPIO Zero version: =1.5.1
Pin factory used: tested with RPiGPIO and Native

The code below is supposed to test pull up/down on a floating pin. It works correctly and outputs `0` and then `1`.

```python
from gpiozero.pins.rpigpio import RPiGPIOFactory
pin = RPiGPIOFactory().pin(5)
pin.function = 'input'
pin.pull = 'down'
print(pin.state)
pin.pull = 'up'
print(pin.state)
```

Changing the factory to `NativeFactory` still works correctly on a Raspberry Pi <4. The code below outputs `False` and then `True`.

```python
from gpiozero.pins.native import NativeFactory
pin = NativeFactory().pin(6)
pin.function = 'input'
pin.pull = 'down'
print(pin.state)
pin.pull = 'up'
print(pin.state)
```
However, on a Pi 4 it will output `True` and `True`. It seems the Pi 4 uses a different pull register and the `NativeFactory` might need to be updated to reflect that.

bennuttall, 2020-10-12:
Yeah, we've got a patch that we haven't got around to testing yet. I'll see if I can dig it out...

lurch, 2020-10-13:
:wave: That's the patch that I made, but haven't looked at for a long long time. I'll try to remember to get around to testing it this weekend :crossed_fingers: 
See also #777

commit 499321a1 (2021-03-02): Add complete Pi4 support to NativePin. Fixes #895 and #777
Also add more comprehensive pull-state tests.