gpiozero/gpiozero #897: Read potentiometer value without a ADC

Ja83La, 2020-10-12:
May it be possible to use SmoothedInputDevice in order to read an approximate potentiometer value without connecting an ADC?

I think of a solution similar to:
https://www.allaboutcircuits.com/projects/building-raspberry-pi-controllers-part-5-reading-analog-data-with-an-rpi/

bennuttall, 2020-10-12:
Sure - see if you can follow that guide replacing RPi.GPIO lines for gpiozero ones. See https://gpiozero.readthedocs.io/en/stable/migrating_from_rpigpio.html

lurch, 2020-10-13:
> https://www.allaboutcircuits.com/projects/building-raspberry-pi-controllers-part-5-reading-analog-data-with-an-rpi/

That's exactly what https://gpiozero.readthedocs.io/en/stable/api_input.html#lightsensor-ldr does :slightly_smiling_face:

closed 2021-02-22 by bennuttall