gpiozero/gpiozero #898: Can't control motors or robot using pigpio

sentairanger, 2020-10-13:
Hello, I am currently trying to control a robot remotely using my Google AIY Voice kit version 2. It successfully was able to blink the LEDs I tested but would not run the motors. I had no issue running the motors locally. I even tried repeating the same issue on my other robot and I still got the same problem. The LEDs work but the robot won't move at all even though the code is correct. Here's the information on all my Raspberry Pis I have tested so far:

Raspberry Pi Zero W/Google Voice Kit Version 2

Operating System: Raspbian Buster
Python Version: 3.7.3
GPIO Zero version 1.5.1
pin factory: RPi.GPIO

Raspberry Pi Zero W/Robot 1(Linus)

Operating System: Raspbian Buster Lite/Raspberry Pi OS Lite
Python Version: 3.7.3
GPIOZero version 1.5.1
Pin Factory: RPi.GPIO

Raspberry Pi 3B+/Robot 2(Torvalds)

Operating System: Raspbian Buster
Python Version: 3.7.3
GPIOZero Version 1.5.1
Pin Factory: RPi.GPIO

I tried running this code using my voice kit to control the robot remotely

```python
#!/usr/bin/env python3
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#
# robot_remote_movement.py
# Moves a robot remotely based on commands
# Modified by Edgardo Peregrino
# 10/4/2020

"""A demo of the Google CloudSpeech recognizer."""
#import libraries
from gpiozero import Robot
from gpiozero.pins.pigpio import PiGPIOFactory
from time import sleep
import aiy.voice.tts

import argparse
import locale
import logging

from aiy.cloudspeech import CloudSpeechClient

#define the factory
factory = PiGPIOFactory(host='192.168.0.17') 
#Define the robot
robot = Robot(left = (4, 17), right = (27, 22), pin_factory=factory)

#create a function that creates hints to what to say to the box
def get_hints(language_code):
    if language_code.startswith('en_'):
        return ('go forward',
                'go backward',
                'go left',
                'go right',
                'goodbye')
    return None

#create a function that defines the local language based on location
def locale_language():
    language, _ = locale.getdefaultlocale()
    return language

def main():
    #log any issues and print the commands on the terminal
    logging.basicConfig(level=logging.DEBUG)
    # use this if you aren't using English
    parser = argparse.ArgumentParser(description='Assistant service example.')
    parser.add_argument('--language', default=locale_language())
    args = parser.parse_args()
    #initialize the cloudspeech API
    logging.info('Initializing for language %s...', args.language)
    hints = get_hints(args.language)
    client = CloudSpeechClient()
    while True:
        #here it detects if your statements match with the hints
        if hints:
            logging.info('Say something, e.g. %s.' % ', '.join(hints))
        else:
            logging.info('Say something.')
        text = client.recognize(language_code=args.language, hint_phrases=hints)
        if text is None:
            logging.info('You said nothing.')
            continue
        logging.info('You said: "%s"' % text)
        # Robot moves based on which ever direction it is told
        text = text.lower()
        if 'go forward' in text:
            aiy.voice.tts.say('Okay I will go forward')
            robot.forward()
            sleep(3)
            robot.stop()
        elif 'go backward' in text:
            aiy.voice.tts.say('Okay I will go backward')
            robot.backward()
            sleep(3)
            robot.stop()
        elif 'go left' in text:
            aiy.voice.tts.say('Okay I will go left')
            robot.left()
            sleep(3)
            robot.stop()
        elif 'go right' in text:
            aiy.voice.tts.say('Okay I will go right')
            robot.right()
            sleep(3)
            robot.stop()
        elif 'goodbye' in text:
            aiy.voice.tts.say('Goodbye')
            robot.stop()
            break

if __name__ == '__main__':
    main()        

```

I even tried the button example from the gpiozero documentation but also had the same problem

bennuttall, 2020-10-13:
@sentairanger what is the smallest python program you can write that replicates the problem? Someone else's 100 line script is difficult to debug. I'm pretty sure as @lurch says it's a problem that's been fixed in master but not yet released. You should be able to replicate it with:

```python
from gpiozero import Motor
from gpiozero.pins.pigpio import PiGPIOFactory

factory = PiGPIOFactory(host='192.168.0.17') 
m = Motor(4, 17, pin_factory=factory)
print(m._all[0].pin_factory)
```

when running on a Pi, it will presumably print `<gpiozero.pins.rpigpio.RPiGPIOFactory...` and when runnning on a PC it will presumably error when trying to create the Motor.

You can fix it by installing from github:

```
sudo pip3 install https://github.com/gpiozero/gpiozero
```

sentairanger, 2020-10-13:
@bennuttall Yes that's the code I ran just now and got an error. Thank you for your help. I thought I did something wrong with my code.

sentairanger, 2020-10-13:
@lurch I didn't think to read those issues but just did. I hope that the next version of gpiozero resolves that issue permanently. I'll try out the fix provided for now. I'll keep this open until there's a more permanent fix.

sentairanger, 2021-02-05:
I am now closing this issue as it now works for me with the following command 

`sudo pip3 install https://bennuttall.com/files/gpiozero-1.5.2b1-py2.py3-none-any.whl` Thanks to you @bennuttall for the suggestion. I can now proceed to adding my servo arm to my robot as planned.

closed 2021-02-05 by sentairanger

lurch, 2021-02-05:
:robot: :mechanical_arm: :+1:

sentairanger, 2021-02-19:
Update, everything worked and I was able to run both of my robots simultaneously.