gpiozero/gpiozero #899: Wait for multiple motion sensors

dieserNiko, 2020-10-22:
Hello there,

I’m currently stuck on a problem and I can’t figure out if I missed something or if I’m really the only one trying to do this. Anyway, I’m trying to couple two motion sensors on two different Pis via remote GPIO. The idea behind this is that I have two motion sensors in my room and I want to turn the light on as soon as one of them detects motion, keep it on as long as at least one detects motion in time periods of 60s or something and turn the light off after both of them detect no motion. 

The integrated function of pir.wait_for_motion seems to only be able to wait for input from one motion sensor, not from multiple though. I also don’t want some kind of unperformant while loop that permanently checks if one of the sensors is active or not. It really should be an event trigger that reacts to both sensors (or the first/last to be triggered). 

Is there any way to do this? I tried to find hints by googling but I didn’t find anything at all and I don’t know where else to ask.

Thanks for your help!

lurch, 2020-10-22:
Sounds like https://gpiozero.readthedocs.io/en/stable/api_tools.html#combining-sources is exactly what you're looking for? :wink:

dieserNiko, 2020-10-23:
Oh wow how could I not have found this? 😅 Thanks, I'll try it out!

dieserNiko, 2020-10-23:
Okay so far so good... I got it until the wait for variable to change... but I don't know how to continue as soon as the variable changes state... any tips? 😅 

> #!/usr/bin/python3
> from gpiozero import LED, MotionSensor
> from gpiozero.pins.pigpio import PiGPIOFactory
> from gpiozero.tools import any_values
> from signal import pause
> 
> schreibtisch = PiGPIOFactory(host='10.0.0.32')
> ventilator = PiGPIOFactory(host='10.0.0.31')
> desk = MotionSensor(20, pin_factory=schreibtisch)  # remote motion sensor
> fan = MotionSensor(17, pin_factory=ventilator)  # remote motion sensor
> 
> motion = any_values(fan, desk)
> 
> while True:
>     #as soon as motion changes state, continue
>     print(motion)
>     print("Motion!")
> 
> pause()

(is this the right way to insert code here? 🤔)

lurch, 2020-10-23:
Best way to insert code is between triple-backticks.

```
#!/usr/bin/python3
from gpiozero import LED, MotionSensor
from gpiozero.pins.pigpio import PiGPIOFactory
from gpiozero.tools import any_values
```

I think if you want to "detect changes" using sourcetools (rather than using the `led.source = any_values(fan, desk)` paradigm), then you would indeed need to resort to using a `while` loop. But as long as you remember to put a `sleep` inside your while loop too (so that the multitasking CPU on the Pi has time to run other tasks too), it won't affect performance.

@bennuttall @waveform80 It's been a long while since I looked at the sourcetools stuff, but I wonder if it'd be possible to write an "edge detection" sourcetool that only yields values when the value changes state from the previous value? Or do sourcetools only work when the number of values being yielded matches the number of values being consumed?

lurch, 2020-10-23:
You may find some of the stuff in #255 interesting too! (I had a feeling this had been discussed before)

dieserNiko, 2020-10-23:
That looks promising, I'll have a look! Thanks!

(Maybe to be sure again, I don't need to trigger a Pin on my pi with the motion sensors because I have to feed the values to HomeBridge to control automations from there... my output has to be some kind of stat file or something that I can check from another script and continue from there later on.)

dieserNiko, 2020-10-28:
Okay so I did it with multiple python an bash scripts feeding into each other... probably not the cleanest way but it works. Thanks for all the help anyway! :)

closed 2020-10-28 by dieserNiko