gpiozero/gpiozero #902: Is it possible to simulate the clicking of a keyboard key using GpioZero?

davidlenehan, 2020-11-18:
I would like to click a momentary switch connected to a GPIO pin and have the Pi recognize it as a click of the letter P (for instance) on a keyboard.

Apologies if this has been explained elsewhere, but I couldn't find it.

lurch, 2020-11-18:
Pulling together a few different strands... there's a gpiozero example [here](https://gpiozero.readthedocs.io/en/stable/recipes.html#keyboard-controlled-robot) that shows using `evdev` to read keypress events from the keyboard... however I believe `evdev` can also be used to simulate events coming from the keyboard, see [here](https://python-evdev.readthedocs.io/en/latest/tutorial.html#injecting-input).

I guess this might make a nice [ButtonBoard](https://gpiozero.readthedocs.io/en/stable/api_boards.html#buttonboard) example! :grinning:

davidlenehan, 2020-11-18:
I have played with your suggestions today. I am trying to do something within RetroPie and EvDev keyboard outputting is weird when you don't have an actual keyboard connected. I don't think it will work for my use case.

Thank you for the help! It set me on the right road.

closed 2020-11-18 by davidlenehan