gpiozero/gpiozero #903: Distance sensor script hangs when sensor is disconnected

foobarchar, 2020-11-20:
I'm using the HC-SR04 ultrasonic sonar distance sensor on a Raspberry Pi zero with Arch Linux ARM. When the sensor is connected the follow code works fine:

#/usr/bin/env/python
import gpiozero
sonar=gpiozero.DistanceSensor(trigger=13,echo=16,queue_len=50,max_distance=1.3)
print(sonar.distance)
print(sonar.max_distance)

and gives the following output:

1.1405518899313796
1.3

which is ok (I am in range of the sensor).

But when the sensor is disconnected (wiring) the code gives the following output and 'hangs':

/usr/lib/python3.8/site-packages/gpiozero/input_devices.py:997: DistanceSensorNoEcho: no echo received
warnings.warn(DistanceSensorNoEcho('no echo received'))

I use this sensor in my robot which is controlled using curses which is continuously reading my sensor outputs and waiting for keyboard input. Sometimes I need to disconnect the distance sensor and test other things but when I do that the curses terminal freezes because of this 'hanging' script.

I tried 'hiding' this warning by using

import warnings
warnings.filterwarnings("ignore")

but the warning keeps coming up and the script just 'freezes'.

I have read the docs, googled and tried to the best of my knowledge but I cannot get rid of the warning. Any ideas?

lurch, 2020-11-20:
Looking at the docs https://gpiozero.readthedocs.io/en/stable/api_input.html#distancesensor-hc-sr04 they say "partial (bool) – When False (the default), the object will not return a value for is_active until the internal queue has filled with values. Only set this to True if you require values immediately after object construction." so _maybe_ when you disconnect the distance sensor it's "hanging" waiting for the queue to fill again? So _perhaps_ setting `partial` to `False` in the constructor will prevent it hanging? (but it'll obviously also lead to more erratic values, as it now won't be averaging them)

foobarchar, 2020-11-20:
@lurch: that did the trick! I modified above code as follows to show what exactly happens:

#!/usr/bin/env python

import gpiozero
import time

sonar=gpiozero.DistanceSensor(trigger=13,echo=16,max_distance=1.5,threshold_distance=0.3,partial=True)

try:
    while True:
        distance=round(sonar.distance*100)
        if distance==0:
            print("    sensor error!",end="\r")
        elif distance<sonar.max_distance*100:
            print(distance)
            time.sleep(1)
        elif distance==sonar.max_distance*100:
            print("    out of range!",end="\r")
            time.sleep(1)

except Exception as error:
    print(error)

finally:
    pass

Output when connected:

111 out of range!
110
112
110
110
    out of range!

Output when disconnected:

/usr/lib/python3.8/site-packages/gpiozero/input_devices.py:997: DistanceSensorNoEcho: no echo received
warnings.warn(DistanceSensorNoEcho('no echo received'))
    sensor error!

The code now keeps running and the "sensor error!" is what shows on my robot's curses GUI. I have to look more closely for any erratic values like you said but it looks ok for now.

Since the issue is solved I close it now but as you can see I still get the pesky warning but it does not affect my robot script anymore.

closed 2020-11-20 by foobarchar