gpiozero/gpiozero #904: gpiozero.exc.PinUnknownPi: unknown old-style revision "0"

presianbg, 2020-11-26:
Operating system:  CentOS 8
Python version:  3.6
Pi model:  Pi 2 Model B rev 1.1
GPIO Zero version: 1.5.1
Pin factory used: RPiGPIO

Hi, 
I'm trying to use this wonderful library on my Pi 2B rev 1.1 with CentOS 8:
`Linux  4.19.72-300.el8.armv7hl #1 SMP Fri Sep 20 17:55:47 UTC 2019 armv7l armv7l armv7l GNU/Linux`
but no matter what `Pin factory` I use, this is the result:

```python
[root@frodo ~]# python3
Python 3.6.8 (default, Nov 21 2019, 22:10:21) 
[GCC 8.3.1 20190507 (Red Hat 8.3.1-4)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero.pins.rpigpio import RPiGPIOFactory
>>> from gpiozero import LED
>>> factory = RPiGPIOFactory()
>>> led = LED(17, pin_factory=factory)
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/data.py", line 1092, in from_revision
    ) = PI_REVISIONS[revision]
KeyError: 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/gpiozero/devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/output_devices.py", line 211, in __init__
    pin, active_high, initial_value, pin_factory=pin_factory
  File "/usr/local/lib/python3.6/site-packages/gpiozero/output_devices.py", line 93, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/mixins.py", line 106, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/devices.py", line 520, in __init__
    self.pin_factory.reserve_pins(self, pin)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/pi.py", line 100, in reserve_pins
    requester, *(self.pi_info.to_gpio(pin) for pin in pins))
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/pi.py", line 100, in <genexpr>
    requester, *(self.pi_info.to_gpio(pin) for pin in pins))
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/__init__.py", line 188, in <lambda>
    lambda self: self._get_pi_info(),
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/pi.py", line 120, in _get_pi_info
    self._info = pi_info(self._get_revision())
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/data.py", line 1365, in pi_info
    return PiBoardInfo.from_revision(revision)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/data.py", line 1094, in from_revision
    raise PinUnknownPi('unknown old-style revision "%x"' % revision)
gpiozero.exc.PinUnknownPi: unknown old-style revision "0"
```

How to debug this issue?

Kind regards,
PY

lurch, 2020-11-26:
Looks like https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/local.py#L93 must be returning `0` on CentOS 8 for some reason. (and `0` isn't a valid [revision code](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md) )
What do `ls -l /proc/device-tree/system/linux,revision` and `cat /proc/cpuinfo` say?

bennuttall, 2020-11-26:
What's your revision code? https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md

Quick fix: edit `/usr/local/lib/python3.6/site-packages/gpiozero/pins/data.py` and add `revision = 'a01041'` after the line `def from_revision(cls, revision):` [here](https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/data.py#L913)

Does gpiozero work for you on Raspbian?

lurch, 2020-11-26:
@bennuttall It's not "Raspbian" any more, it's "Raspberry Pi OS" :wink: https://www.raspberrypi.org/software/

presianbg, 2020-11-26:
@lurch:

cat /proc/cpuinfo
```
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5
```
`ls: cannot access '/proc/device-tree/system/linux,revision': No such file or directory`
but:

```
cat /sys/firmware/devicetree/base/model
Raspberry Pi 2 Model B
```
@bennuttall  - should be `a01041` indeed. I will try your workaround.
This Pi was not used for few years, but decided to bring it back to life yesterday. 
Haven't used Raspbian before, but OSMC which is debian based too, but didn't tried this library with it....
I was using `gpio` binary to power on / off some leds / fans,  which is not available in CentOS.

presianbg, 2020-11-26:
@bennuttall with the workaround I have another exception:
```
Python 3.6.8 (default, Nov 21 2019, 22:10:21) 
[GCC 8.3.1 20190507 (Red Hat 8.3.1-4)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero.pins.rpigpio import RPiGPIOFactory
>>> from gpiozero import LED
>>> factory = RPiGPIOFactory()
>>> led = LED(17, pin_factory=factory)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/gpiozero/devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/output_devices.py", line 211, in __init__
    pin, active_high, initial_value, pin_factory=pin_factory
  File "/usr/local/lib/python3.6/site-packages/gpiozero/output_devices.py", line 93, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/mixins.py", line 106, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/devices.py", line 520, in __init__
    self.pin_factory.reserve_pins(self, pin)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/pi.py", line 100, in reserve_pins
    requester, *(self.pi_info.to_gpio(pin) for pin in pins))
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/pi.py", line 100, in <genexpr>
    requester, *(self.pi_info.to_gpio(pin) for pin in pins))
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/__init__.py", line 188, in <lambda>
    lambda self: self._get_pi_info(),
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/pi.py", line 120, in _get_pi_info
    self._info = pi_info(self._get_revision())
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/data.py", line 1366, in pi_info
    return PiBoardInfo.from_revision(revision)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/data.py", line 915, in from_revision
    if revision & 0x800000:
TypeError: unsupported operand type(s) for &: 'str' and 'int'
```

lurch, 2020-11-26:
Hmmm, that's weird - if `/proc/device-tree/system/linux,revision` doesn't exist, and none of the lines in `/proc/cpuinfo` start with `Revision` then it seems https://github.com/gpiozero/gpiozero/blob/v1.5.1/gpiozero/pins/local.py should `raise PinUnknownPi('unable to locate Pi revision in /proc/device-tree or /proc/cpuinfo')`, rather than returning `0` and causing `raise PinUnknownPi('unknown old-style revision "%x"' % revision)` to happen at a later step? :shrug: 

To fix the workaround, try changing `revision = 'a01041'` to `revision = 0xa01041` ?

lurch, 2020-11-26:
Is the `vcgencmd` command available in CentOS?

presianbg, 2020-11-26:
> Is the `vcgencmd` command available in CentOS?
Do not think so... Couldn't find the package:
`dnf provides "*/vcgencmd"`
`Error: No Matches found`

Ok, we have some progress here:
```
>>> from gpiozero.pins.rpigpio import RPiGPIOFactory
>>> from gpiozero import LED
>>> factory = RPiGPIOFactory()
>>> led = LED(17, pin_factory=factory)
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/pi.py", line 109, in pin
    pin = self.pins[n]
KeyError: 17

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/gpiozero/devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/output_devices.py", line 211, in __init__
    pin, active_high, initial_value, pin_factory=pin_factory
  File "/usr/local/lib/python3.6/site-packages/gpiozero/output_devices.py", line 93, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/mixins.py", line 106, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/devices.py", line 521, in __init__
    pin = self.pin_factory.pin(pin)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/pi.py", line 111, in pin
    pin = self.pin_class(self, n)
  File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/rpigpio.py", line 132, in __init__
    GPIO.setup(self.number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Mmap of GPIO registers failed
```
what else beside the `revision` is broken :(

lurch, 2020-11-26:
```
File "/usr/local/lib/python3.6/site-packages/gpiozero/pins/rpigpio.py", line 132, in __init__
    GPIO.setup(self.number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Mmap of GPIO registers failed
```

Looks like RPi.GPIO itself isn't working on CentOS either? :confused: 
If you search through some of the Ubuntu-related issues https://github.com/gpiozero/gpiozero/search?q=ubuntu&type=issues you _may_ find some pointers on things to try, to get it working? (or you could see if you get any better results with a different [Pin Library](https://gpiozero.readthedocs.io/en/stable/api_pins.html#changing-the-pin-factory)?)

Is there a reason you're committed to using CentOS, or would using Raspberry Pi OS be acceptable (where we know that RPi.GPIO and GpioZero definitely work :wink: )

presianbg, 2020-11-27:
Looks like OS bug. Closing.

closed 2020-11-27 by presianbg