gpiozero/gpiozero #908: MCP3208

mekano69, 2020-12-20:
Operating system: Raspbian Buster
Python version: e.g. 3.8
Pi model: e.g. Pi 3+ Model B
GPIO Zero version: 1.5.1
Pin factory used:  <gpiozero.pins.rpigpio.RPiGPIOFactory object at 0x76a9b9d0>

I'm doing a project with the RaspberryPi and Python where I have 32 relays, and I'm doing 3 measures for each relay, then i made the mean of 3 measures and store in one list.
The measurement works great, but sometimes a error occurs and blocks my entire GUI, the reason - > ADC blocks when its suppost to measure and I dont know why !

This is the code line where it blocks.
```python
leitura=gpiozero.MCP3208(channel=canal[ncanal], differential=False, max_voltage=adc_max_volt_ref, port=0, device=0).voltage
```

An image of the error:
[image]

bennuttall, 2021-02-22:
Have you tried installing `spidev` and seeing if that makes a difference?

mekano69, 2021-02-22:
Of course, I already have installed that before that error

[image]

This is the 3 functions that i create to make my measures, and today still blocks occur in the same line I said in the first question:
```python
    def average(self,lst):
        var = sum (lst)/len(lst)    
        self.vdc_media=var
        
    def leitura_adc(self,ncanal, adc_max_volt_ref, casas_decimais):
        vect_leituras=[]
        canal={1:0, 2:1, 3:2, 4:3, 5:4, 6:5, 7:6, 8:7}
        leitura=0
        cnt=0
        
        for i in range(3): # vai fazer X mediçoes
            while leitura==0:
                try:
                    print("leitura antes: ",leitura)
                    time.sleep(0.3)
                    leitura=gpiozero.MCP3208(channel=canal[ncanal], differential=False, max_voltage=adc_max_volt_ref, port=0, device=0).voltage # clock_pin=11, mosi_pin=10, miso_pin=9, select_pin=8
                    print("leitura depois: ",leitura)
                except:
                    leitura=0    
                    cnt=cnt+1
                    if cnt == 5:
                        string="Verificar ligação do ADC ao sistema!!!"
                        funcoes.func_imprime_gui(self,string,1)
                        break
            if cnt==5:
                print("Verificar ligação do ADC ao sistema!!!")
                break
            else:
                vect_leituras.append(leitura) # mete num vetor
                leitura=0
        if cnt==5:
            self.resultado_leitura_adc=0 # NAO DETETOU O ADC
        else:
            self.average(vect_leituras) # faz a media das 3 leituras
            self.resultado_leitura_adc=str(round(self.vdc_media,casas_decimais))

    def verifica_medicao(self,ncanal, adc_max_volt_ref, casas_decimais, valor_nominal, valor_min, valor_max):
        self.leitura_adc_signal.emit(ncanal, adc_max_volt_ref, casas_decimais) # NAO SE ENVIA O self
        ler_contato=self.resultado_leitura_adc
        if ler_contato==0 or ler_contato==0.0: 
            string="Verificar ligação do ADC ao sistema!!!  medição: "+str(ler_contato)      
            funcoes.func_imprime_gui(self,string,1)
            self.resultado_verifica_medicao=2              # NAO DETETOU O ADC
        else:
            ler_contato=float(ler_contato)
            if ler_contato < valor_nominal:
                if ler_contato < valor_min:
                    string="Componente MAU ... Medição: "+str(ler_contato)+"V"
                    funcoes.func_imprime_gui(self,string,1)
                    self.resultado_verifica_medicao=0
                else:
                    string="Componente BOM ... Medição: "+str(ler_contato)+"V"
                    funcoes.func_imprime_gui(self,string,1)
                    self.resultado_verifica_medicao=1
            if ler_contato > valor_nominal:
                if ler_contato < valor_max:
                    string="Componente BOM ... Medição: "+str(ler_contato)+"V"
                    funcoes.func_imprime_gui(self,string,1)
                    self.resultado_verifica_medicao=1
                else:
                    string="Componente MAU ... Medição: "+str(ler_contato)+"V"
                    funcoes.func_imprime_gui(self,string,1)
                    self.resultado_verifica_medicao=0
```

waveform80, 2021-03-15:
Your code is repeatedly creating and destroying the MCP3208 device. Given the amount of locking that goes under the covers to try and ensure no simultaneous access to ports, it's likely you're running into a race condition somewhere with a non-re-entrant lock trying to be acquired twice by the same thread.

That is a bug we should look into (might actually be fixed in 1.6.0 - there was quite a lot of re-working of the SPI layer in that), but in the meantime you can almost certainly work around it by just constructing the MCP3208 device once and re-using it. For example, somewhere in your initialization:

```python3
    def __init__(self, max_volt, *args):
        self.adcs = [
            gpiozero.MCP3208(channel=canal, differential=False,
                max_voltage=max_volt, port=0, device=0)
            for canal in range(8)]
        # All the rest of your init...
```

Then to read them:

```python3
        for i in range(3): # vai fazer X mediçoes
            while leitura==0:
                try:
                    print("leitura antes: ",leitura)
                    time.sleep(0.3)
                    leitura=self.adcs[canal[ncanal]].voltage
                    print("leitura depois: ",leitura)
                except:
                    leitura=0    
                    cnt=cnt+1
                    if cnt == 5:
                        string="Verificar ligação do ADC ao sistema!!!"
                        funcoes.func_imprime_gui(self,string,1)
                        break
```

mekano69, 2021-04-11:
Thank you very much @bennuttall your solution solve my problem :D

closed 2021-04-11 by mekano69