gpiozero/gpiozero #910: OutputDevice.Value only outputs 0

CoderLD-LX, 2020-12-30:
Operating system: Raspbian GNU/Linux 10 (buster)
Python version: 3
Pi model: Pi 4 Model B Rev 1.1
GPIO Zero version: 1.5.1
Pin factory used: RPiGPIO

So I went to install the pi-fan-controller created by user josephtyler and noticed some weird behavior. I proceeded to create some sanity check print statements to debug the code and found that the .value call was only returning 0 even when the value of the entity was 1.

I thought maybe it was a weird thing with the .on() call and so I tried assigning the .value to 0 and 1 and it had the desired effect of turning the GPIO pin on and off but I was still seeing 0 in all of the places that I was having the code print status reports.

the code below is the modified version of josephtyler's code that I tried to get some feedback from. In all of those checks (even the one immediately after the fan.on() statement) I would only see a 0 in the terminal feedback.

```python
#!/usr/bin/env python3

import subprocess
import time

from gpiozero import OutputDevice

ON_THRESHOLD = 60  # (degrees Celsius) Fan kicks on at this temperature.
OFF_THRESHOLD = 45  # (degress Celsius) Fan shuts off at this temperature.
SLEEP_INTERVAL = 5  # (seconds) How often we check the core temperature.
GPIO_PIN = 14  # Which GPIO pin you're using to control the fan.
FANSTATUS = ["OFF","ON"]

fan = OutputDevice(GPIO_PIN)

def get_temp():
	"""Get the core temperature.
	Run a shell script to get the core temp and parse the output.
	Raises:
	RuntimeError: if response cannot be parsed.
	Returns:
	float: The core temperature in degrees Celsius.
	"""
	output = subprocess.run(['vcgencmd', 'measure_temp'], capture_output=True)
	temp_str = output.stdout.decode()
	try:
		return float(temp_str.split('=')[1].split('\'')[0])
	except (IndexError, ValueError):
		raise RuntimeError('Could not parse temperature output.')

if __name__ == '__main__':
	# Validate the on and off thresholds
	if OFF_THRESHOLD >= ON_THRESHOLD:
		raise RuntimeError('OFF_THRESHOLD must be less than ON_THRESHOLD')

	while True:

		# Grab the CPU temperature
		temp = get_temp()

		print("\n\n\nTemperature = " + str(temp) + "\nOn Threshold = " + str(ON_THRESHOLD) + "\nOff Threshold = " + str(OFF_THRESHOLD) + "\nFan Status = " + FANSTATUS[fan.value])

		# Start the fan if the temperature has reached the limit
		# and the fan is off
		# NOTE: `fan.value` returns 1 for "on" and 0 for "off"
		if temp > ON_THRESHOLD and not fan.value:
			fan.on()
			print("\nEntered Fan ON (" + str(fan.value) + ")")

		# Stop the fan if it is on and the temperature has dropped
		# below lower limit
		elif fan.value and temp < OFF_THRESHOLD:
			fan.off()
			print("\nEntered Fan OFF (" + str(fan.value) + ")")

		time.sleep(SLEEP_INTERVAL)
```

Again this code is not mine, I simply altered some comments and added some debugging for myself. josephtyler has shared this through both GitHub and HowChoo.

I did search in both the open issues for something along these lines and did not find anything so apologies if I missed it and am beating a dead horse with this post.

bennuttall, 2020-12-30:
I'm not sure what you're referring to when you say `.value` is wrong.

There's a lot of code there that's doing a very simple task which can be achieved much easier using the built-in `CPUTemperature` class:

```
from gpiozero import OutputDevice, CPUTemperature

fan = OutputDevice(14)
cpu = CPUTemperature()

while True:
    if cpu.temperature > 45:
        fan.on()
    elif cpu.temperature > 60:
        fan.off()
```

Some debugging you can do, open a Python shell, paste the first few lines (not the while loop) and type things like:

```
>>> cpu.temperature
>>> fan.value
>>> fan.on()
>>> fan.value
>>> fan.off()
>>> fan.value
```

See https://gpiozero.readthedocs.io/en/latest/api_internal.html#cputemperature

CoderLD-LX, 2020-12-30:
@bennuttall I'm very new at programming in Linux/Raspbian and subsequently with the gpiozero library so I trusted the code I saw as it was working to obtain the CPU temp. That being said, it did seem inefficient although I'm not experienced enough to be able to know a simpler path than the one that was given to me so thank you for the simpler method within the gpiozero library.

[image]

doing the debugging method you recommended, I got the same output that I was getting yesterday. The fan is turning on but the .value variable of the fan object reports 0

bennuttall, 2020-12-30:
That's odd.

Does that value ever change?

```python
while True:
    fan.toggle()
    print(fan.value)
```

I'd expect alternating 0/1. If not, something else must be controlling the pin. Do you have another program running that's controlling the pin?

CoderLD-LX, 2020-12-31:
So this is a hardware error and not software. I ran a test to see if I could get fan.value to work on another GPIO on my machine and that worked so then I began thinking since I'm fairly confident that I didn't have anything other than this script messing with GPIO 14, then I might be chasing the symptom not the cause.

The reason I kept getting 0 was because the voltage on that GPIO pin was being pulled down by the transistor that was switching on the fan. This in turn was not allowing `fan.toggle()` to function properly or the code that was provided to function as intended. 

Apologies for jumping to conclusions last night, I was simply too tired from head-to-keyboard bashing to properly troubleshoot.

Thanks again for the code optimization help @bennuttall at the very least I learned something from this ordeal 😅

closed 2020-12-31 by CoderLD-LX