gpiozero/gpiozero #911: InputDevices Problem (Button on GPIO27)

Koloblix, 2020-12-30:
Operating system: Ubuntu Mate 20.10
Python version: e.g. 3.8.6
Pi model: Pi 4 Model B 4gb
GPIO Zero version: 1.5.1
Pin factory used: RPiGPIO

Using simple Python Script playing sounds on button pressed. Just works fine on all button, but on GPIO27.

at first my Script:

```python
from gpiozero import Button
import pygame.mixer
from pygame.mixer import Sound
from signal import pause

pygame.mixer.init()

button_sounds = {
    Button(4): Sound("/home/user/bla1.wav"),
    Button(17): Sound("/home/user/bla2.wav"),
    Button(22): Sound ("/home/user/bla3.wav"),
    Button(27): Sound ("/home/user/bla4.wav"),
}
for button, sound in button_sounds.items():
    button.when_pressed = sound.play
    print(button, "pressed")  

pause()
```
@print(button, "pressed")  it prints in shell:
```python
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
<gpiozero.Button object on pin GPIO4, pull_up=True, is_active=True> pressed
<gpiozero.Button object on pin GPIO17, pull_up=True, is_active=True> pressed
<gpiozero.Button object on pin GPIO22, pull_up=True, is_active=True> pressed
<gpiozero.Button object on pin GPIO27, pull_up=True, is_active=False> pressed
```
and on GPIO27 is_active=False 
Using "Button(27, active_state=True)

makes this error:
```python
gpiozero.exc.PinInvalidState: Pin 27 is not floating, but "active_state" is not None
```

bennuttall, 2020-12-30:
Did you set the pin state outside of Python?

I don't think you want to use `active_state`. I think you want `pull_up=bool` e.g `Button(27, pull_up=False)`

Koloblix, 2020-12-30:
No, havent set the pin state outside of python. Buttons are pulled down with 10k to GND, i dont need internal pull up/down.

Just wondering why GPIO27 is the only button, which dosnt work, and print says "is_active=FALSE"

closed 2020-12-30 by Koloblix

lurch, 2020-12-31:
If you're using external pull-down resistors, you probably want e.g. `Button(27, pull_up=None, active_state=True)`