gpiozero/gpiozero #913: Difference between Button.is_pressed and Button.when_pressed

DonJuape, 2021-01-07:
What is the difference between Button.is_pressed and Button.when_pressed?\
Currently I am using Button.when_pressed in my (first) project: https://github.com/DonJuape/EnergyMeter and it works but I wonder if Button.is_pressed is maybe more reliable in my case? I am counting 100ms long pulses and not missing any pulses is much more important then the exact time of the start of the pulse.

lurch, 2021-01-07:
They do different things
 - `.is_pressed` tells you whether the button is currently being pressed when you call it
 - `.when_pressed` allows you to assign an event-handler / callback-function that gets called each time the button is pressed

Have a read through the very comprehensive documentation:
https://gpiozero.readthedocs.io/en/stable/recipes.html
https://gpiozero.readthedocs.io/en/stable/api_input.html#button

> maybe more reliable

Is there a specific problem you're having?

DonJuape, 2021-01-08:
Thanks, I now understand how they are different. From the documentation I did not get this.
Is it correct to say that with .is_pressed I have to do the polling myself in the script and that with .when_pressed the script waits for an interrupt (in this case a rising edge) on the GPIO input pin?
I am not having a specific problem but I want to minimize the chance that I miss a pulse (i.e. miss a button press in the gpiozero sense).

lurch, 2021-01-08:
> Is it correct to say that with .is_pressed I have to do the polling myself in the script and that with .when_pressed the script waits for an interrupt (in this case a rising edge) on the GPIO input pin?

That's correct :+1: 
Tracing back through the GpioZero code `.when_pressed` eventually boils down to using https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/rpigpio.py#L247 which is documented here https://sourceforge.net/p/raspberry-gpio-python/wiki/Inputs/

closed 2021-01-08 by DonJuape