gpiozero/gpiozero #914: Add support for 5 pins rotary encoder

courtiol, 2021-01-08:
Device name: e.g. Rotary Encoder
Protocol: GPIO
URL of example product: https://www.berrybase.de/en/components/passive-components/potentiometer/rotary-encoder/drehregler/rotary-encoder-mit-breakoutboard

Many projects, e.g. sound box, rely on a rotary encoder; e.g. https://blog.sharedove.com/adisjugo/index.php/2020/05/10/using-ky-040-rotary-encoder-on-raspberry-pi-to-control-volume/

It seems that there has been a PR ready for some times #482
It would be great if it could be merged.

Many thanks!

bennuttall, 2021-02-22:
Is there a big difference between this device and the one #482 is made for? It looks like maybe 3 pins (plus power/ground) rather than 2?

courtiol, 2021-02-23:
They are the same (the extra pin adds a push button).
I wrote this issue precisely to motivate the merge of #482 into master :-)

bennuttall, 2021-02-23:
If this issue is no different to #392 then I'll close as duplicate

closed 2021-02-23 by bennuttall