gpiozero/gpiozero #915: `MockFactory` instances not independent?

pmlk, 2021-01-23:
Operating system: Raspbian Buster / MacOS 11
Python version: 3.9
Pi model: Raspberry Pi Model B Plus Rev 1.2
GPIO Zero version: 1.5.1
Pin factory used: MockFactory / PiGPIOFactory

## Issue

It seems to be impossible to create two independent `MockFactory` instances. 

## Expectations

* be able to create multiple devices using the same pins with different instances of a `MockFactory`
* each `MockFactory` instance mocks a single, independent RasPi

## Notes

Interestingly, this works just fine with two instances of `PiGPIOFactory` even when they control the same pins on the same physical RasPi.

## Details

I would expect both these tests to pass (by not throwing any exceptions):

```python
import time

from gpiozero import DigitalOutputDevice
from gpiozero.pins.mock import MockFactory
from gpiozero.pins.pigpio import PiGPIOFactory

pin_number = 23

def test_multiple_factories_same_pin():
    pi_dns = "raspberrypi.local"
    factory_1 = PiGPIOFactory(host=pi_dns)
    factory_2 = PiGPIOFactory(host=pi_dns)

    led_1 = DigitalOutputDevice(pin_number, pin_factory=factory_1)
    led_2 = DigitalOutputDevice(pin_number, pin_factory=factory_2)

    led_1.on()
    time.sleep(1)
    led_2.off()

def test_multiple_mock_factories_same_pin():
    factory_1 = MockFactory()
    factory_2 = MockFactory()

    led_1 = DigitalOutputDevice(pin_number, pin_factory=factory_1)
    led_2 = DigitalOutputDevice(pin_number, pin_factory=factory_2)

    led_1.on()
    led_2.off()

if __name__ == "__main__":
    test_multiple_factories_same_pin()
    test_multiple_mock_factories_same_pin()
```

`test_multiple_mock_factories_same_pin` fails with `gpiozero.exc.GPIOPinInUse: pin 23 is already in use by <gpiozero.DigitalOutputDevice object on pin GPIO23, active_high=True, is_active=False>`.

I had a quick look at the code and added some print statements after looking at the abstract `Factory` in `gpiozero/pins/__init__.py`:

```python
    print(factory_1._reservations)
    print(factory_2._reservations)

    led_1 = DigitalOutputDevice(pin_number, pin_factory=factory_1)

    print(factory_1._reservations)
    print(factory_2._reservations)

    led_2 = DigitalOutputDevice(pin_number, pin_factory=factory_2)
```

The printed dictionaries for the two instances are the same: 

```python
defaultdict(<class 'list'>, {})
defaultdict(<class 'list'>, {})
defaultdict(<class 'list'>, {23: [<weakref at 0x10571d950; to 'DigitalOutputDevice' at 0x105723460>]})
defaultdict(<class 'list'>, {23: [<weakref at 0x10571d950; to 'DigitalOutputDevice' at 0x105723460>]})
```

So, somehow the `_reservations` attribute seems to be synchronized between all `MockFactory` instances, causing the above error. Is this intentional?

**Why do I want/expect this behavior (independent `MockFactory` instances)?**  
When I write tests in different modules, I don't want to have to keep track of the already used pins when I instantiate new devices with an all new `MockFactory`. Also, I could (theoretically) run out of pins for all tests and then I would need to manage the release of pins before/after tests are executed.

P.S.: If my expectation is, in fact, the desired behavior, I'd be willing to submit a PR. So far, I've only been a (recent) user of this great library, so I may need some guidance.

bennuttall, 2021-01-24:
Interesting.

1. It feels like a bug that multiple PiGPIOFactory instances to the same address can be used without collision, although I'm not sure how we'd detect that if e.g. you created one with IPv4 and one with IPv6 or internal/external IPs - but it might be nice to notice two identical IPs and not allow it. Not a huge problem but hey.

2. It seems reasonable to want independent Mock factories. Funnily enough, I previously suggested the opposite - an alternative mock factory with persistence between programs (so you could run multiple programs hitting the same pins) - and also a graphical mock pin. So there's scope for multiple mock factories - we definitely need the current one as we use it in the test suite - but an alternative factory, or an alternative mode (set with an init param) would be useful.

@waveform80 and I are busy with other things at the moment, so if you want to have a go, take a look at https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/mock.py - the implementation of mock pins. And read up on pin factories in the docs here: https://gpiozero.readthedocs.io/en/latest/api_pins.html

pmlk, 2021-01-27:
Thanks, I'll have a look this weekend.

From what you're saying, I get the feeling that I have a fundamentally different expectation of how things would work under the hood. So I expect to have some questions once I get a closer look.

waveform80, 2021-03-15:
There's a hierarchy to the factory/pins side of things that goes roughly like this:

* `Factory` / `Pin` - largely abstract base classes that make no assumptions about what machine or pins they represent. Might be a pi, might not. Might be general purpose pins, or pins restricted to input / output duties. Might provide hardware-driven SPI interfaces, might not.
* `PiFactory` / `PiPin` - Derivative of the above that represents GPIO pins on *a* Raspberry Pi. I emphasize *a* because they're not necessarily attached to the Pi you're running on - could be a remote one, but either way they're definitely on a Raspberry Pi.
* `LocalPiFactory` / `LocalPiPin` - Derivative of the above that represents GPIO pins on *this* Raspberry Pi - i.e. the one you're running your code on.

As you go down the hierarchy more and more bits are implemented. For example, `LocalPiFactory` has methods to determine the revision code of the local pi by asking the file-system for various things. `PiFactory` can't do that because the Pi in question is not necessarily the same as the local file system's.

`MockFactory` (and `MockPin` et al) descend from the `LocalPiFactory` level. Why? Because it was easiest layer to do so from and was sufficient for the device tests, which was the reason for creating the mock pins stuff in the first place (to be able to test the device logic on non-pi hardware like CI systems).

The `LocalPiFactory` includes some special logic to prevent people from instantiating multiple local pi factories, e.g. by trying to get `RPIOFactory` and `RPiGPIOFactory` going side by side, because multiple drivers of a single GPIO system is not something we handle (or particular want to handle for that matter).

`MockFactory`, by dint of descending from `LocalPiFactory`, inherits this block. It *could*, with a little work, descend from `PiFactory` instead. But that wasn't needed for running our device tests, so I took the lazy route. Still, it's an interesting idea, and probably a useful chunk of functionality for some people. @bennuttall stick this on the TODO list?

In the meantime, PRs welcome!

pmlk, 2021-03-16:
Thanks for the very thorough explanation :)

> In the meantime, PRs welcome!

Here you go: #919 :)

waveform80, 2021-03-19:
> Thanks for the very thorough explanation :)
> 
> > In the meantime, PRs welcome!
> 
> Here you go: #919 :)

Unfortunately that's not really the approach I was talking about  - that fixes the issue by adding a parameter to MockFactory without changing the fact that it descends from LocalPiFactory when what should happen is the whole lot should move up to PiFactory. I've had a quick look at the code and the majority of what LocalPiFactory adds is stuff to do with revision detection ... which MockFactory doesn't care about (because we just tell it what revision we want), so no loss there. It *would* meant the duplication of the ticks code but that's trivial (in fact I've just noticed I can ditch that comment about monotonic being time because that was a 2.x compat. thing anyway).

It'd also be an opportunity to clean up the process by which MockFactory provides SPI interfaces - that's all a bit muddy at the moment but really ought to be another construction parameter like pin_class is (so a mock SPI device, like the mocks of the MCP3xxx ADCs could be passed on construction during those tests). Anyway, now I've had a skim of the code it doesn't look like a whole lot of work - I'll have a stab at it this weekend.

pmlk, 2021-03-19:
> Unfortunately that's not really the approach I was talking about 

As I noted in the PR, I am aware that my fix is more of hack than a good, clean solution. At the time, I didn't feel comfortable restructuring the class hierarchy.

I appreciate you tackling this :) Feel free to close my PR.

waveform80, 2021-03-21:
Just posted #946 which should deal with this as cleanly as I can manage - seems to pass the test originally proposed at the top.

Just to address one final point I note I haven't dealt with:

> Interesting.
>
> 1. It feels like a bug that multiple PiGPIOFactory instances to the same address can be used without collision, although I'm not sure how we'd detect that if e.g. you created one with [...]

I wish we could but (as you've probably guessed) it's basically impossible for all intents and purposes, and I don't like beating my head against unsolvable problems :)

commit f0d1b71e (2021-04-04): Move SPI tests

Many of the SPI tests need to be moved under the "real" pins tests as
now that MockFactory descends from PiFactory rather than LocalPiFactory,
most of the logic handling software vs hardware SPI implementations
won't run otherwise.

This is a good opportunity to test hardware SPI initialization properly,
but that does imply that the "regular" pin tests mustn't be sitting on
the SPI pins. While we're at it, might as well ban the I2C pins from
this too in case we get around to adding that interface.

Those tests that don't need to move to real pins, go to test_spi_devices
instead [... cut]
closed 2021-04-04 by waveform80