gpiozero/gpiozero #916: Show and control mock pins

KUGA2, 2021-01-24:
I program something on my desktop with gpiozero.

The mock library works (it it starts) but i do not know if it switches Pins correctly.
How can I see the stated of the pins gpiozero controls?

Can I also send input Pins, like simulating the push of a button?

bennuttall, 2021-01-24:
Yes, you can emulate a button being pressed like so:

```python
>>> from gpiozero import LED, Button
>>> from gpiozero.pins.mock import MockFactory
>>> mock = MockFactory()
>>> led = LED(10)  # real led
>>> btn = Button(11, pin_factory=mock)  # mock button
>>> led.source = btn  # this means the LED state matches the button
>>> led.value  # the led is initially off
0
>>> btn.pin.drive_low()  # mock the button being pressed
>>> led.value  # the led is now on because the button is pressed
1
>>> btn.pin.drive_high()  # mock the button being released
>>> led.value  # the led is now off because the button is not pressed
0
```

If you want more advanced examples, refer to how we test devices in the test suite: https://github.com/gpiozero/gpiozero/tree/master/tests

Hope that helps

KUGA2, 2021-01-24:
Well yes, i know this.

But when i run the script with my logic, i cannot modify its variables.  Or can I? I confess, i do not know how to use the python shell very deeply.

bennuttall, 2021-01-24:
Do you want to be able to run a script that does the same thing every time, like wait 5 seconds then emulate the button being pressed, or do you want to be able to run the script then e.g. press the spacebar to emulate the button being pressed?

KUGA2, 2021-01-26:
I **have** a script that does something with pins:
e.g. when button is pressed, toggle LED. run forever.

I want so see and control the these pins from another process. E.g. from another (python) shell.

Ideally you would have a graphical interface (web interface?) showing a pinout, what pin is controlled by gpiozero, to what direction it is set, its state and, when it is input, you can click to drive high/low.

bennuttall, 2021-01-27:
Yeah, it would be nice. We're not actively working on the library at the moment, busy with other things, so I wouldn't wait for us to build it.

If I were you I'd mock something up with tk/guizero - have an on-screen button that acts as a replacement trigger instead of a GPIO button. Could do the same with a small web API using e.g. Flask.

bennuttall, 2021-03-12:
Closing this as a duplicate of #445

closed 2021-03-12 by bennuttall