gpiozero/gpiozero #918: Wrong LED selected when turning on a named pin by index

Teraskull, 2021-01-26:
Operating system: e.g. Raspberry Pi OS
Python version: 3.7.1
Pi model: Pi 3 Model B+
GPIO Zero version: 1.5.1

When creating an LEDBoard with named pins and accessing one of them by index, the order is wrong:

```python
leds = LEDBoard(red=17, yellow=27, green=22)

leds.on(0)  # This will turn on green instead of expected red.
```

```python
leds = LEDBoard(17, 27, 22)

leds.on(0)  # This will turn on expected red.
```

I believe this is because the order gets alphabetically sorted, but is not intuitive.

lurch, 2021-01-26:
That's because `(red=17, yellow=27, green=22)` is effectively a dictionary, and dictionaries have no inherent ordering (which is why they get alphabetically sorted, so at least they have _some_ form of consistent ordering).
I believe you could `print(leds.leds)` to see what order the indices actually end up as? But I think the expectation is that if you're specifying leds by name during construction, then you'd also access them by name when using them? e.g. `leds.red.on()`

Oh! But it looks like specifying the `_order` (as described at https://gpiozero.readthedocs.io/en/stable/api_boards.html#compositeoutputdevice ) _might_ also work for LEDBoards? (if it does, I guess this is a gap in the documentation)

bennuttall, 2021-01-26:
If you use the TrafficLights class, it's ordered correctly:

```python
In [8]: tl = TrafficLights(red=22, yellow=23, green=24)

In [9]: tl.on(0)

In [10]: tl.value
Out[10]: TrafficLightsValue(red=1, yellow=0, green=0)
```

But as @lurch said, keyword arguments are unordered, so you need to use `_order`:

```python
leds = LEDBoard(red=17, yellow=27, green=22, _order=('red', 'yellow', 'green'))
```

Teraskull, 2021-01-26:
Oh, okay, _order works for me, thanks. I just wondered if this is related to dictionary ordering from Python 3.6

closed 2021-01-26 by bennuttall

lurch, 2021-01-28:
@bennuttall I guess `_order` needs to be added to the docs here https://gpiozero.readthedocs.io/en/master/api_boards.html#gpiozero.LEDBoard & https://gpiozero.readthedocs.io/en/master/api_boards.html#gpiozero.LEDCollection

(Normally I'd create a PR for that myself, but I'm kinda busy dealing with Raspberry Pi Pico issues at the moment!)

bennuttall, 2021-01-28:
Yeah, there are a few things like that missing from the docs I'd like to add at some point. I learned recently how to include docs for magic methods too (so we can document __iter__ for LEDBoard.

waveform80, 2021-03-20:
> Oh, okay, _order works for me, thanks. I just wondered if this is related to dictionary ordering from Python 3.6

You are correct that once 3.6 becomes our "base" version, the `_order` keyword will be effectively redundant and could be removed (technically Python 3.7 would be the base for this as that's when the spec changed, but AFAIK there's no Python 3.6 implementations with unordered dicts). However, while we've now dropped 2.7 support for a future gpiozero 2.0 release, the new base is currently 3.5 to support Debian/Raspbian oldstable (stretch).

@bennuttall If bullseye gets released prior to gpiozero 2.0 we could (should) look at removing `_order` but if we release first, it'll have to stay until 3.0 as it'd a breaking change to remove it.

lurch, 2021-03-21:
Perhaps my brain is just too hardwired to the old Python2 ways of thinking :brain: , but personally I always like to treat `dict` as an inherently unordered type; and even on Python3.6+ I still explicitly use `OrderedDict` whenever I want an ordered dict, as IMHO it makes the code-intention clearer :shrug: 
So IMHO I'd be :-1: on the removal of the `_order` keyword. (of course the other advantage of a separate `_order` keyword is that you can explicitly give an order which may be different to the dict's order)

bennuttall, 2021-03-21:
It would be nice to remove `_order`. I don't think there's any advantage to releasing 2.0 until we have sufficient _new_ features (worthy of a 1.7 release on their own), so let's see. Certainly if we release after Bullseye I'd be happy to remove it.

waveform80, 2021-03-21:
> Perhaps my brain is just too hardwired to the old Python2 ways of thinking brain , but personally I always like to treat `dict` as an inherently unordered type;

I share your preferences for thinking of dicts as unordered (personally I always thought that fitted very neatly with the `keys()` method returning a view which acted as a set, and could be treated as such -- and of course, even in 3.6+ sets are still unordered too). But, the language moves on, c'est la vie (also, the reasons why they're now ordered and yet more efficient as a result are quite interesting - there's a Raymond H talk somewhere that goes through the evolution of the dict over time which is worth a look).

> and even on Python3.6+ I still explicitly use `OrderedDict` whenever I want an ordered dict, as IMHO it makes the code-intention clearer shrug

Actually very sensible. Despite dicts being ordered by default, OrderedDict is still a separate type with subtle differences in semantics and performance. For instance:

```python3
>>> from collections import OrderedDict
>>> regular_dict = {'a': 1, 'b': 2}
>>> regular_dict == {'b': 2, 'a': 1}
True
>>> ordered_dict = OrderedDict([('a', 1), ('b', 2)])
>>> ordered_dict == OrderedDict([('b', 2), ('a', 1)])
False
```

In other words, while both types preserve order, only OrderedDict cares about the order for the purposes of equality comparison. It's a bit like the difference between case sensitive file-systems vs case retentive, but insensitive file-systems; dict preserves the order but doesn't "care" about it. They also have subtle differences in performance (dict is better for iteration, but OrderedDict is better for re-ordering).

Anyway, let's sit on the fence about this for now. I'll have to do a bit of digging to see whether the ordered nature of dict is liable to be reverted in the future. IIRC, the only reason it came about was because it was a side-effect of the more efficient storage algorithm -- if that's the case and another, better algorithm comes along there's no reason for that behaviour to persist. Although it would require a major version bump for them to revert it, so it's highly unlikely.