gpiozero/gpiozero #921: Why force active_state = False when pull_up = True?

pmlk, 2021-02-12:
My question is regarding the following couple of lines:

https://github.com/gpiozero/gpiozero/blob/a6f7efeb749045b4da7519e6abbc03309ee4a0f9/gpiozero/input_devices.py#L112-L124

I'd like to understand the reasoning behind this seemingly very deliberate design choice. From a few examples I remember seeing, it seems to force the following assumption (when `pull_up=True`): A button is connected to a pin and GND; when the button is pressed, the pin is electrically LOW (connected to GND) but logically the pin's state should be HIGH (or "active"), when the button is released the pin is pulled electrically HIGH (to avoid a floating pin), but should be logically LOW.

This makes sense. But why assume and force this scenario when `pull_up=True`? Why not allow `pull_up=True` and `active_state=True`?

Let me offer my perspective and my current use case:

I have a water reservoir with two sensors, one near the top to signal a "full" state and one near the bottom to signal an "empty" state. Basically, when there is water the sensors close a connection, when there is not enough water the connection is open.

I would like to connect the "Empty Sensor" to a pin and have the pin be logically HIGH when there is no more water. To me the most obvious way is to simply enable a pull up resistor for the pin and have the logical state be analogous to the electrical state. As long as there is water in the reservoir the sensor closes the connection between the pin and GND (electric == logic == LOW), once the reservoir is empty, the connection between the pin and GND is open, but the pin is pulled up to an electrical as well as logical HIGH.

Of course, there are a number of ways to get around this. I would just like to understand the reasoning behind this. 

Also, I'd be happy to open a PR to allow `pull_up=True and active_state=True`.

pmlk, 2021-02-18:
Also just noticed that passing `pull_up=True` and `active_state=False` throws a `PinInvalidState` Exception (`'Pin %d is not floating, but "active_state" is not None'`). To me this is even more astonishing since I am just explicitly passing what is otherwise automatically assumed anyways.

I would propose the following: 

1. Only when `active_state is None` should `active_state` be set automatically based on `pull_up`.
2. When both `active_state is None` and `pull_up is None` ("floating"), set `active_state=True` as a sensible default, rather than throwing an exception.

OR even more radical

3. decouple these two options entirely:
    * set sensible defaults (e.g. `pull_up=None`<sup>[1](#footnote1)</sup>, `active_state=True`)
    * never make an assumption about one option based on the other, assume the user knows what their doing

So it's either **1. and *maybe* 2.** OR **only 3.** I think 3. would be the best, most straightforward option, but I assume it would break some tests.

<span>____</span>
<sup><a name="myfootnote1">1</a></sup>: MCP23017 (see #651) doesn't support pull-down resistors which breaks the current default `pull_up=False` and forces one to specify `pull_up=None` or `pull_up=True` upon pin or device instantiation.

cinderblock, 2021-06-26:
This is a major issue for us. Our hardware is a button that is normally closed, which shorts an IO pin to `GND`. So `pull_up=True` is required for our hardware.

However, the *logic* of our button is that it is **pressed** when the IO pin goes to a `HIGH` level (because the normally closed button stops shorting to `GND` and the `pull_up` biases the IO pin to `HIGH`.

Basically, IMHO, the exceptions raised [here](https://github.com/gpiozero/gpiozero/blob/f203f4abe0e11c5f82ef952370ddebca5a02bf0c/gpiozero/input_devices.py#L83-L94) just get in the way.

If you *need* to keep this exception for some compatibility reason, an extra `bool` argument could override this behavior.

lurch, 2021-06-28:
> Also just noticed that passing pull_up=True and active_state=False throws a PinInvalidState Exception ('Pin %d is not floating, but "active_state" is not None'). To me this is even more astonishing since I am just explicitly passing what is otherwise automatically assumed anyways.

@pmlk IIRC the current "API" is that `pull_up` is specified and `active_state` is left as `None` (and `active_state` is calculated automatically from `pull_up`), **OR** `pull_up` is None (i.e. external pull-up/down resistors are being used), and an `active_state` is explicitly passed. See #658 for more background.

> never make an assumption about one option based on the other, assume the user knows what their doing

GpioZero was explicitly designed to be beginner-friendly, which is why it makes more assumptions than the lower-level pin libraries that it builds on top of.

> This is a major issue for us. Our hardware is a button that is normally closed, which shorts an IO pin to GND. So pull_up=True is required for our hardware.

@cinderblock I guess the current logic was designed around normally-open buttons, and didn't take (unusual?) normally-closed buttons into consideration? As a workaround (until @waveform80 weaves his magic) I guess you could use an external pull-up resistor and then pass `pull_up=None, active_state=True` ?

cinderblock, 2021-06-28:
Yeah, this "button" is a safety button. If any of them is broken, whole machine needs to stop. So it needs to be normally closed.

You're right that I could add an extra pull-up myself but I would prefer to not and don't see a reason to do this since the Pi hardware can handle it.

I was considering using `InputDevice` directly but it turns out these extra exceptions are thrown from `InputDevice`. Maybe we could just move these "easy mode" exceptions to the `Button` class and let the `InputDevice` class let you shoot yourself in the foot?

Makman2, 2023-10-28:
I hit the same limitation, and I agree that the exception is rather artificialy. Normally-closed are as common as normally-open switches, so I think they should be definitely supported by this without throwing any exceptions. Sensible defaults like they are not could stay in place to keep backwards-compatibility.

I would submit a PR that addresses the issue with your first suggestion @pmlk :

> Only when active_state is None should active_state be set automatically based on pull_up.

This is as backwards-compatible as possible and also conveys a common assumption that I experienced in general that a button is usually normally-open.

The second suggestion

> When both active_state is None and pull_up is None ("floating"), set active_state=True as a sensible default, rather than throwing an exception.

would also be good imo. But I think for simplicity this could be moved into a different issue to keep patch submissions smaller.

Some other arguments to remove the exception:

From @lurch (https://github.com/gpiozero/gpiozero/issues/921#issuecomment-869588453):

> @pmlk IIRC the current "API" is that pull_up is specified and active_state is left as None (and active_state is calculated automatically from pull_up), OR pull_up is None (i.e. external pull-up/down [...]

I think you meant specifically https://github.com/gpiozero/gpiozero/pull/658#discussion_r243788744. There is no real discussion why this is good. The change proposal was just accepted. To our vain now :sweat_smile: 

> GpioZero was explicitly designed to be beginner-friendly, which is why it makes more assumptions than the lower-level pin libraries that it builds on top of.

I agree this is important. But I think allowing managing the active state yourself is not beginner-unfriendly but just required for the different types of switches that exist. Especially there will be good defaults (i.e. the same as before) in place.

> @cinderblock I guess the current logic was designed around normally-open buttons, and didn't take (unusual?) normally-closed buttons into consideration? As a workaround (until @waveform80 weaves his [...]

Normally-closed buttons are not uncommon from my experience. And being able to avoid an external resistor just speeds things up during development.

E.g. I have currently an experimental setup where I want to avoid extra circuitry because I would need to solder or tack another breadboard onto the Raspberry Pi I develop the experiment application on. Just stucking jumper cables makes it much easier :smile:

Makman2, 2023-10-28:
Created #1099.