gpiozero/gpiozero #923: gpiozero functions with threading not working on Windows Python?

toddk63, 2021-02-18:
Operating system: e.g. Windows 10
Python version: e.g. 3.7
Pi model: Pi Zero W
GPIO Zero version: 1.5.1
Pin factory used: PiGPIOFactory

I want to control GPIO remotely over my LAN from a Win 10 machine with Python, GPIOZero and pigpio installed. The following works when executed C:\python blink.py :

```
from gpiozero import LED
from gpiozero.pins.pigpio import PiGPIOFactory
from time import sleep

factory = PiGPIOFactory(host='192.168.0.10')
led = LED(17, pin_factory=factory)

led.blink(1,1,None,False)
```

The following does not work:

```
from gpiozero import LED
from gpiozero.pins.pigpio import PiGPIOFactory
from time import sleep

factory = PiGPIOFactory(host='192.168.0.10')
led = LED(17, pin_factory=factory)

led.blink()
```
So it appears functions using threading do not work from Windows command line.

The error given is the error given:

C:\>python blink.py
Exception in thread Thread-2:
Traceback (most recent call last):
File "C:\Users\AppData\Local\Programs\Thonny\lib\threading.py", line 926, in _bootstrap_inner
self.run()
File "C:\Users\AppData\Local\Programs\Thonny\lib\threading.py", line 870, in run
self._target(*self._args, **self._kwargs)
File "C:\Users\AppData\Roaming\Python\Python37\site-packages\gpiozero\output_devices.py", line 275, in _blink_device
self._write(True)
File "C:\Users\AppData\Roaming\Python\Python37\site-packages\gpiozero\output_devices.py", line 106, in _write
self.pin.state = self._value_to_state(value)
File "C:\Users\AppData\Roaming\Python\Python37\site-packages\gpiozero\pins\__init__.py", line 305, in <lambda>
lambda self, value: self._set_state(value),
File "C:\Users\AppData\Roaming\Python\Python37\site-packages\gpiozero\pins\pigpio.py", line 267, in _set_state
elif self.function == 'input':
File "C:\Users\AppData\Roaming\Python\Python37\site-packages\gpiozero\pins\__init__.py", line 283, in <lambda>
lambda self: self._get_function(),
File "C:\Users\AppData\Roaming\Python\Python37\site-packages\gpiozero\pins\pigpio.py", line 240, in _get_function return self.GPIO_FUNCTION_NAMES[self.factory.connection.get_mode(self.number)]
File "C:\Users\AppData\Roaming\Python\Python37\site-packages\pigpio.py", line 1402, in get_mode
return _u2i(_pigpio_command(self.sl, _PI_CMD_MODEG, gpio, 0))
File "C:\Users\AppData\Roaming\Python\Python37\site-packages\pigpio.py", line 1026, in _pigpio_command
dummy, res = struct.unpack('12sI', sl.s.recv(_SOCK_CMD_LEN))
AttributeError: 'NoneType' object has no attribute 'recv'

If I run this on Thonny (installed on Windows 10), it works!. I have even tried deleting the Windows Python installation and setting the Environment PATH to the Thonny Python installation. I get similar results with other functions that use threading i.e. pulse().

bennuttall, 2021-02-22:
Using Thonny's Python sounds a bit dodgy to me. However the error boils down to the pigpio library.

I would suggest trying to do something similar using the pigpio library directly. See if you can reproduce the error using pigpio without threads.

@joan2937 does the above error mean anything to you?

joan2937, 2021-02-22:
The difference between the two calls appears to be whether background is true or false.

`blink(on_time=1, off_time=1, n=None, background=True)`

Perhaps the pigpio default should be False (it makes no sense to pigpio as the daemon does not spawn tasks for PWM).

bennuttall, 2021-02-22:
Well the threading is handled by gpiozero, which would (in a thread) send pigpio commands to control the PWM. So if it is reproducible in gpiozero but not by doing the same thing in pigpio alone then it's a bug in gpiozero. I'll test on Windows when I get chance.

toddk63, 2021-02-23:
Thanks.  if I understand correctly, the benefit to having threading is 
that it can do its infinite loop (blink) while continuing with the 
program execution that follows.

Todd K.

waveform80, 2021-03-15:
Your second script (ending with the bare `led.blink()`) will terminate immediately when run from the command line because there's nothing following the blink call. The script ending will shut down the background pigpio connection which would lead to an error like the one you've observed. That won't happen in thonny because the interpreter's still running.

Can you try adding `from signal import pause` and `pause` to the end of your script?

toddk63, 2021-03-15:
>Can you try adding `from signal import pause` and `pause` to the end of your script?

Great suggestion. Apparently signal.pause is NOT supported in Window.  A search of the text in signal.py did not find "pause"

waveform80, 2021-03-15:
> > Can you try adding `from signal import pause` and `pause` to the end of your script?
> 
> Great suggestion. Apparently signal.pause is NOT supported in Window. A search of the text in signal.py did not find "pause"

Argh, of course it isn't! Sorry, it's so rare I deal with Windows these days, I've forgotten a lot of the compatibility bits. Erm - let me see, what's a suitable alternative ... I guess at it's most basic an infinite loop of sleep's ought to suffice. Something like:

```python
from time import sleep

# ... rest of the script goes here ...

while True:
    sleep(1)
```

All you want it to do is keep the main thread alive while background threads do their thing, but without the main thread pegging a CPU busy (while a trivial `while True: pass` loop would do). There's probably a better way of doing this under Windows but I can't find anything useful in the signal module.