gpiozero/gpiozero #926: Cython compiled simple button code raises gpiozero.exc.BadEventHandler

nixternal, 2021-02-23:
Operating system: Raspbian GNU/Linux 10 (buster)
Python version: 2.7.16
Pi model: Raspberry Pi 4 Model B Rev 1.1
GPIO Zero version: 1.5.1
Pin factory used: gpiozero.pins.rpigpio.RPiGPIOFactory

```python
#!/usr/bin/env python
# cython: language_level=2

from gpiozero import Button
from signal import pause

def pressed(btn):
    print('GPIO {} Pressed'.format(btn.pin.number))
    return

button = Button(26)
button.when_pressed = pressed
pause()
```

Compiled with Cython using:

```sh
cython -o bld/"${1}".c --embed src/"${1}".py
clang -c bld/"${1}".c -o bld/"${1}".o -O2 -I/usr/include/python2.7
clang -o ${pub}/"${1}" bld/"${1}".o -lpython2.7
```

And here is the error I get when running the compiled code:

```
Traceback (most recent call last):
  File "src/tester.py", line 14, in init tester
  File "/usr/lib/python2.7/dist-packages/gpiozero/devices.py", line 148, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python2.7/dist-packages/gpiozero/mixins.py", line 251, in when_activated
    self._when_activated = self._wrap_callback(value)
  File "/usr/lib/python2.7/dist-packages/gpiozero/mixins.py", line 338, in _wrap_callback
    'value must be a callable which accepts up to one '
gpiozero.exc.BadEventHandler: value must be a callable which accepts up to one mandatory parameter
```

bennuttall, 2021-02-24:
While that looks simple, I'm not surprised it doesn't work with cython. We don't support cython, and can't really help here beyond suggesting that you modify the source code to remove the callable check and see if you can get it to work. Alternatively, try another activation method, e.g. a while loop and `if button.is_pressed` if you can.

nixternal, 2021-02-24:
I'll try that later, I've switched to using RPi.GPIO & add_event_detect in the mean time. Thanks for the quick response!

bennuttall, 2021-02-25:
cool, closing this but if you have any questions feel free to ask

closed 2021-02-25 by bennuttall

waveform80, 2021-03-14:
Sorry I didn't notice this earlier -- the problem in this case is *probably* that your Cython function takes a parameter. I know we promise in the docs that functions for event handlers can take 0 or 1 parameters but that's a slight white lie. *Python* functions given as event handlers can have an optional parameter, but stuff that's externally compiled (including Python built-ins and thus Cython stuff too) cannot: they can only have zero parameters.

The programming reason for this is that we do this 0 or 1 param magic by inspecting the function we're given. However, you can't inspect externally compiled or built-in functions - you just have to "know" how many parameters they take.

That then begs the question: why did go for 0 parameters in the default case rather than 1? The rationale here is that if we went for one parameter, that parameter will be a gpiozero object. That assumes the thing we're calling therefore "knows" about gpiozero. If we're talking about something from the standard library (e.g. Event.set) then it can't know about gpiozero, and so by demanding it takes a gpiozero-esque parameter we deny the ability to use it.

Assuming I'm right about the cause here, you've got two options:

1. Re-write your handler to take zero parameters

2. Wrap your handler with something like functools.partial to pass in the thing that's calling it (which is basically all we do when we get a python handler with 1 param anyway - see EventsMixin._wrap_callback in gpiozero/mixins.py for the gory details).