gpiozero/gpiozero #947: CPUTemperature not firing callbacks

TheBum, 2021-03-23:
I'm trying to use gpiozero.CPUTemperature to monitor CPU temperature and fire off when_activated and when_deactivated callbacks to turn a PWM fan on and off, respectively.  However, despite the temperature crossing the threshold, the callbacks are not firing. I'm running this in Python 3.7.3 on a Raspberry Pi Zero W. Here's some test code I'm attempting to use. The "PWM" line is a simple 0%/100% duty cycle signal; I'm not interested in running any other speeds.

    import time
    import pigpio
    import gpiozero

    RUN_TIME = 300.0
    SAMPLE_TIME = 2.0

    def start_fan():
            print('Starting fan')
            pi.write(25,1)

    def stop_fan():
            print('Stopping fan')
            pi.write(25,0)

    pi = pigpio.pi()

    p = gpiozero.CPUTemperature(threshold=30.0)
    p.when_activated = start_fan
    p.when_deactivated = stop_fan
    if p.is_active:
            start_fan()
    else:
            stop_fan()

    start = time.time()

    while (time.time() - start) < RUN_TIME:

            time.sleep(SAMPLE_TIME)
            print('threshold = {0}, temperature = {1}, is_active = {2}'.format(p.threshold, p.temperature, p.is_active))

    p.cancel()

    pi.stop()

I can see the `is_active` state changing as the threshold is crossed, but no fan control changes.

bennuttall, 2021-03-23:
We only just implemented this in the new release - you must have version 1.6.x - I don't think it's in the apt repo yet but you can pip install it.

FYI you can do the above without pigpio e.g:

```
from gpiozero import OutputDevice, CPUTemperature
from signal import pause

cpu = CPUTemperature(threshold=30)
fan = OutputDevice(25)

cpu.when_activated = fan.on
cpu.when_deactivated = fan.off

pause()
```

TheBum, 2021-03-23:
Yeah, I had 1.5.1. However, I'm unable to update using pip3 because of a certificate verification error:

    Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Max retries exceeded with url: /packages/c0/fa/fd051ec230fb841476a626662e31f580756811ad70392f3ee6ae56ad52a9/gpiozero-1.6.2-py2.py3-none-any.whl (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1056)')))

waveform80, 2021-03-23:
> Yeah, I had 1.5.1. However, I'm unable to update using pip3 because of a certificate verification error:
> 
> ```
> Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Max retries exceeded with url: /packages/c0/fa/fd051ec230fb841476a626662e31f580756811ad70392f3ee6ae56ad52a9/gpiozero-1.6.2-py2.py3-none-any.whl (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1056)')))
> ```
> 
> EDIT: I was able to download the wheel file and pip3 install that.

That verification error is because your pi's clock hadn't yet synced its time. Note that it states "certificate is not yet valid" - that indicates that the certificate's start date is in the future (at least as far as your pi is concerned); it's a common problem when attempting anything involved HTTPS before NTP has synced (or on networks where NTP is being suppressed for some reason or other).

closed 2021-05-27 by bennuttall