gpiozero/gpiozero #950: ssh connection lost when controlling onboard LEDs

lole-elol, 2021-04-09:
Operating system: Raspbian Buster
Python version: e.g. 3.7
Pi model: Pi 4 Model B Rev 1.4
GPIO Zero version: 1.5.1
Pin factory used: RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

I am connected to my pi via ssh.

When i run the above script also mentioned in the [documentation](https://gpiozero.readthedocs.io/en/stable/recipes_advanced.html#controlling-the-pi-s-own-leds), my ssh session colses and it takes 2 to 4 minutes until the pi is reachable via ssh again.

```python
from gpiozero import LED
from signal import pause

 
power = LED(35) # /sys/class/leds/led1
activity = LED(47) # /sys/class/leds/led0

activity.blink()
power.blink()
pause()
```

bennuttall, 2021-04-10:
The on-board LEDs are not accessible this way on the Pi 4. The notes below explain the differences between models and this bit:

> On the Pi 3 the LEDs are controlled by a GPIO expander which is not accessible from gpiozero (yet).

also applies to the Pi 4 - sorry that's not mentioned.

lole-elol, 2021-04-10:
So is there any different way i can controll the two onboard LEDs with python ?
I dont want do da anything fancy just turning them off and on (no blinking etc.) it has to be in python though.

bennuttall, 2021-04-10:
Yes but you'll need a way to access the GPIO expander - I haven't done it myself. I'm sure somebody else has. Try the forums, stack overflow, etc. Some links here: https://gpiozero.readthedocs.io/en/stable/#issues-and-questions

closed 2021-04-10 by bennuttall