gpiozero/gpiozero #951: I want to run a program in which when button is pressed the led blinks 10 times with a delay and if the button is kept pressed the led while stay on until i dont rlease the button.

harryyx, 2021-04-11:
Im new learner and All i know so far is:

from gpiozero import LED, Button
from time import sleep
from signal import pause

led = LED(7)
btn = Button(25)

def light():
    led.on()
    sleep(1)
    led.off()
    sleep(1)

btn.when_pressed = light

pause()

lurch, 2021-04-11:
Sounds like you want to do _something_ like:

```python
def turn_on():
    led.on()

def blink_ten_times():
    led.blink(n=10, background=False)
    led.off()

btn.when_pressed = turn_on
btn.when_released = blink_ten_times

pause()
```
(note that this is just typed in ad-hoc, I've not actually tried running it)

Note also that you'll need to debounce your button-press (either in hardware or software), because you don't want the switch-bounce that happens when you initially press the button to be misinterpreted as a button-release.

harryyx, 2021-04-11:
Oh yes it is something like that i fixed it and all is left to add the delay in this and I'm all done. Thankyou so much for the idea 🙏🏼🙏🏼🙏🏼

closed 2021-04-11 by bennuttall