gpiozero/gpiozero #953: Why strip the debian directory?

m-roberts, 2021-04-12:
[This](https://github.com/gpiozero/gpiozero/commit/627da93d7f39edca3646e7538fd00fe1915c4bd6) commit removes the `debian/` directory.

The commit mentions:
> This should be done with separate branches off release tags.

However, I'm not sure why this is necessary or beneficial. This seems entirely non-standard, and just obfuscates how the package is being built for others.

Is it something to do with worrying about the additional burden of changelog management?

I'd be happy to discuss this further, as I think that it's worth keeping in the codebase!

waveform80, 2021-04-12:
Sorry -- I haven't gotten around to uploading the branch with the replacement in it yet, but I'm also reasonably convinced that the way I've been dealing with debian packaging historically was "not good". It shouldn't be packaged as a native package -- indeed it's much easier to deal with local variation when it's in the quilt format (I've been prepping an updated package for Ubuntu hirsute, and that was one of the major drivers behind this given we want lgpio as the primary backend there instead of rpi.gpio due to the kernel changes).

Going forward the master branch'll be where development happens (as before). At release, we'll add the usual tag and upload that to PyPI (as before). Off that tag will be a branch called something like "debian/1.6" which'll add the debian directory back in and which can grow to accommodate "downstream" patches (gpiozero-1.6-1, gpiozero-1.6-2, etc. as needed).

When we do a new release, I'll cherry-pick that branch onto the new tag and update the changelog for the new release (that part's always been a manual operation anyway so no major changes there). Depending on whether Ubuntu diverges from Debian on this (at the moment it will because I missed the cut-off for Bullseye, but hopefully I can get the changes upstream next cycle), I may add "ubuntu/1.6" branches and the like -- already got those locally but I rather assume nobody much cares about those so I haven't pushed them.

waveform80, 2021-04-12:
I've just pushed the raspios/1.6.2 branch (which branches off from the v1.6.2) which contains the `debian/` directory that I'm intending to use for the 1.6.2 packaging for raspios (I haven't called it debian/1.6.2 as I'm not sure it's what Debian are going to want upstream -- there's complications given that Debian hasn't got the colorzero dependency yet and I'm not sure they'll want a Python 2.7 package for bullseye).

Anyway, that should give you an idea of what the cleaned up packaging looks like:

* Switches to the 3.0 quilt format
* Places the docs under `/usr/share/doc/python3-gpiozero/html` with symlink from `/usr/share/doc/python-gpiozero-doc/html`
* Builds docs with local intersphinx object inventories for build environments with limited/no internet access
* Runs test suite properly during the package build for both 2.7 and 3.x packages (but excludes "real" pin tests for obvious reasons)
* Includes an appropriate `debian/watch` file

Once we release 2.0, there'll be a v2.0.0 tag, and a debian/2.0 branch off that (assuming I've got things upstream "properly" at that point) and so on. Hope that's good enough for your needs?

FWIW, I always upload source packages to raspios anyway, so you can always grab the debian packaging from archive.raspberrypi.org too.